RetroPie/RetroPie-Setup #1936: uinput works fine in RetroPie / EmulationStation but fails when going into game.

codecardinal, 2017-03-24:
I've successfully gotten uinput to emulate keyboards keys when a GPIO pin is active.
This works great when in the EmulationStation menu, but once a game is loaded, it fails to read the emulated keys.

codecardinal, 2017-03-25:
I have a few hunches to go off, in case anyone ever has the same problem.

RetroArch uses libudev by default (https://github.com/libretro/RetroArch/wiki/Input-drivers-in-Linux-without-Xorg)

Now, it might just be that RetroArch only looks for devices in `/dev/input/`. 
The uinput is located at `/dev/uinput`, I could try to write a udev rule that aliases this.

Another thing I will try is to swap the input_driver from udev to sdl or linuxraw.

codecardinal, 2017-03-25:
I've added the following udev rule to symlink it, but that did not work. It allow non root user to access the virtual device and creates a symlink in /dev/input. Did not solve the issues.

`KERNEL=="uinput", SUBSYSTEM=="misc", MODE:="0666", SYMLINK+="input/event1"`

I've also tried switching up driver to SDL, linuxraw etc but that did not help.
I have also tried hardcoding the input_driver to be udev.

joolswills, 2017-03-25:
This is not a RetroPie issue - if you think you have a RetroArch bug you can report it to their project, or for development discussion, please use the forum as indicated when opening an issue.

>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

>For ideas and enhancements, please also post to the forum first, to see if it is something we would consider and to see if other users would find it useful.

>Please remove this text before posting.

closed 2017-03-25 by joolswills