RetroPie/RetroPie-Setup #1937: Dispmanx unusble in 4.2 build

dukesilverfan, 2017-03-25:
I posted in the forum this week about some video stuttering I was having, and together we worked it out to be related to the dispmanx video driver in the 4.2 retropie build, at least on the raspberry pi 2/3.

I am able to reproduce this behavior on a clean install of 4.2 onto my Pi 3 and using the grid scrolling test in 240p test suite in snes9x2010. When video_driver is left to the default 'unset', everything works perfectly, but when you change it to 'dispmanx', the video stutters and lags, making games unplayable. By toggling between 'unset' and 'dispmanx', you can make it perfect, then stutter, perfect, then stutter.

On a clean install of the 4.1 build, both 'unset' and 'dispmanx' work perfectly, no stutter at all, which makes me believe it is directly related to a bug in the 4.2 release.

joolswills, 2017-03-25:
You didn't follow the instructions when opening an issue

>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

In this case, it may well be related to dispmanx driver in RetroArch - which has had some changes. I think the triple buffering was removed. However, for issues with RetroArch you should use their forum or bugtracker.

joolswills, 2017-03-25:
I realise you have discussed this on the forum, but the idea is when something is confirmed as a retropie issue we can move it here. This is related to RetroArch.

closed 2017-03-25 by joolswills

dukesilverfan, 2017-03-25:
Sorry, it wasn't that I didn't mean to follow the rules, I saw that other dispmanx issues were put here, and solved, so I felt it was a correct place, but I see that is wrong now. Now that I know it is related to RetroArch, I will open up an issue there, and hopefully they can coordinate a fix.

dukesilverfan, 2017-04-18:
For documentation sake, this issue has appeared to be solved (or at least mostly solved), noted in: https://github.com/libretro/RetroArch/pull/4621 by commit: https://github.com/RetroPie/RetroPie-Setup/commit/cfaa04786b885b19c6246d15d72836a0e4270a03