RetroPie/RetroPie-Setup #1940: add wip/experimental libretro vice

joolswills, 2017-03-27:
as title. 

I have a WIP module already.

joolswills, 2017-03-27:
preliminary module added to experimental. The core is a bit WIP still, and lots of stuff isn't hooked up to the GUI. The gui also doesn't seem to save any settings, and by default "True drive emulation" is off, which breaks d64 loading.

Recommend people use the standalone vice emulator for now.

closed 2017-03-27 by joolswills

windale, 2017-03-28:
When "True drive emulation" is OFF you have to enable "Virtual device traps". The difference in loading speed with TDE off is night and day, but like you say, the settings don't save anyway and it's very WIP.

joolswills, 2017-03-28:
Issues should be reported to the libretro core. They are aware of some of the problems though.

commit c11760a3 (2017-03-27): added lr-vice to experimental - vice port for libretro. fixes #1940