RetroPie/RetroPie-Setup #1941: Retropie B button (Snes)

d3m0n3y3z, 2017-03-27:
Beforehand I would like to thank any and all kinds of help.

Im having an issue with the B button configured in my retropie. While playing as an example Mario, whenever I press and hold the B button, it only does a short jump and cant hold any shells while holding the button. I have a BBC controller configured in Retroarch and in EmulationStation both. The weird thing is, if I try pressing the button on the keyboard, it works fine.

joolswills, 2017-03-27:
Did you not read the text you removed when posting ?

>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

>For ideas and enhancements, please also post to the forum first, to see if it is something we would consider and to see if other users would find it useful.

closed 2017-03-27 by joolswills

d3m0n3y3z, 2017-03-27:
thanx anyways....