RetroPie/RetroPie-Setup #1947: [Request] cdogs-sdl

Omnija, 2017-04-10:
I've compiled and tested the game and has been working. I think it would a nice addition to the ports games and hopefully gets incorporated. Has controller and multiplayer support.

Source
https://github.com/cxong/cdogs-sdl

Prerequisites
https://github.com/cxong/cdogs-sdl/wiki/Developer-Getting-Started:-Linux

Screenshots of Proof
http://i.imgur.com/sUWwNTN.jpg
http://i.imgur.com/mPMSq4K.jpg

joolswills, 2017-04-10:
We prefer requests to go to the forum first as detailed when opening an issue

>For ideas and enhancements, please also post to the forum first, to see if it is something we would consider and to see if other users would find it useful.

>Please remove this text before posting.

zerojay, 2017-07-22:
With this merged, shouldn't this issue be closed?

joolswills, 2017-07-22:
yep. Thanks.

closed 2017-07-22 by joolswills