RetroPie/RetroPie-Setup #1955: Configuring N64 Controller USB Adapter fails in inputconfiguration.sh

schaal, 2017-04-20:
Trying to configure an USB Adapter for N64 Controllers, identified as `0079:1879 DragonRise Inc.` by lsusb, fails. The following error is printed several times to the console:

```
input config finish command:
/home/pi/.emulationstation/es_temporaryinput.cfg:3.36: xmlParseCharRef: invalid xmlChar value 24
    <inputConfig type="joystick" deviceName="&#24;USB GamePad USB GamePad" deviceGU
```

The problem appears to be, that the adapter has an invalid character in the deviceName, which can also be seen in the [output of udevadm monitor](https://github.com/RetroPie/RetroPie-Setup/files/942504/udev.txt) when attaching the adapter:

```
KERNEL[15869.043710] add      /devices/pci0000:00/0000:00:10.0/usb3/3-1/3-1.4/3-1.4:1.0/0003:0079:1879.001E (hid)
ACTION=add
DEVPATH=/devices/pci0000:00/0000:00:10.0/usb3/3-1/3-1.4/3-1.4:1.0/0003:0079:1879.001E
HID_ID=0003:00000079:00001879
HID_NAME=USB GamePad USB GamePad
HID_PHYS=usb-0000:00:10.0-1.4/input0
HID_UNIQ=
MODALIAS=hid:b0003g0001v00000079p00001879
SEQNUM=3083
SUBSYSTEM=hid

KERNEL[15869.044378] add      /devices/pci0000:00/0000:00:10.0/usb3/3-1/3-1.4/3-1.4:1.0/0003:0079:1879.001E/input/input44 (input)
ABS=30027
ACTION=add
DEVPATH=/devices/pci0000:00/0000:00:10.0/usb3/3-1/3-1.4/3-1.4:1.0/0003:0079:1879.001E/input/input44
EV=1b
KEY=ffff00000000 0 0 0 0
MODALIAS=input:b0003v0079p1879e0110-e0,1,3,4,k120,121,122,123,124,125,126,127,128,129,12A,12B,12C,12D,12E,12F,ra0,1,2,5,10,11,m4,lsfw
MSC=10
NAME="USB GamePad USB GamePad"
PHYS="usb-0000:00:10.0-1.4/input0"
PRODUCT=3/79/1879/110
PROP=0
SEQNUM=3084
SUBSYSTEM=input
UNIQ=""
```

The full es_temporaryinput.cfg:

[es_temporaryinput.cfg](https://github.com/RetroPie/RetroPie-Setup/files/942533/es_temporaryinput.txt)

joolswills, 2017-04-20:
Not sure if I can work around this (might be possible via some changes to the way ES saves the XML)  - but you should report it to the vendor - as it looks like they should remove that character from the device name.

joolswills, 2017-04-20:
For now you can just configure retroarch via RGUI.

joolswills, 2017-04-21:
xmstarlet is not going to be happy parsing it from reading the docs. The only thing I can think of is to save the name mime64 encoding in the xml or extract the name separately from xmlstartlet (via grep), and then clean the file before parsing it. 

Right now it's easiest just to manually adjust your configuration - it's not a common issue thankfully.

Gerii, 2017-04-24:
@schaal Did you have any luck setting up mupen64plus with it? I just got the same (broken) adapter from Amazon.

schaal, 2017-04-25:
@Gerii Unfortunately not

Gerii, 2017-04-25:
@schaal Did you also order it at Amazon.de?

schaal, 2017-04-25:
@Gerii Yes, I also got it from amazon

Langerz82, 2020-10-19:
Try in:
RetroPie-Setup\scriptmodules\supplementary\emulationstation\inputconfiguration.sh

line: 
DEVICE_NAME=$(xmlstarlet sel --text -t -v "/inputList/inputConfig/@deviceName" "$es_conf")
replace:
DEVICE_NAME=$(xmlstarlet sel --text -t -v "/inputList/inputConfig/@deviceName" "$es_conf" | sed 's/&/&amp;/g')

I'm not positive will work but try and test as I'm on win comp atm.

3rd3, 2024-07-16:
Not a bug fix, but I did find a somewhat hacky workaround for this. The problem appears to be that the device count and name get stuck after Mupen64 encounters the first device with the malformed name containing unicode garbage. The solution is to edit the automatically adapted Mupen64 config file after launching it once, such that the `device` number increases and also set the mode for all controllers to manual. The next time it will work correctly, but of course it will not auto-detect any controllers anymore. If you have multiple joysticks or adapters, make sure to connect the bad one first.