RetroPie/RetroPie-Setup #1959: Bad mapping on n64 (mupen)

stevenwalton, 2017-04-23:
I have two controllers bought off of Amazon. One [wired
](https://www.amazon.com/gp/product/B00KBAE52S/ref=oh_aui_search_detailpage?ie=UTF8&psc=1) the other [bluetooth](https://www.amazon.com/Retro-8Bitdo-Wireless-Bluetooth-Controller-Android/dp/B01M1KP9GT/ref=cm_rdp_product). I noticed the wired one had a few bad maps but was quickly fixed (forgot which ones, because I just fixed it and called it a day). On the new bluetooth one It was just atrocious. Input using the "Configure Input" menu in the main menu has
```
D_Up = axis 1-
D_Down = axis 1+
D_L = axis 0-
D_R = axis 0+
Start = 11
A = 0
B = 1
L-Shoulder = 6
R-Shoulder = 7
L-Trigger (Z) = 19
Up Analogue = Hat 0 up
Down = Hat 0 down
Left = Hat 0 left
Right = Hat 0 right
Right Analogue Up = 8
down = 3
left = 4
right = 9
```
And checking `/opt/retropie/configs/n64/InputAutoCfg.ini` after I find I get 
```
; 8Bitdo N64 Gamepad_START
[8Bitdo N64 Gamepad]
plugged = True
plugin = 2
mouse = False
AnalogDeadzone = 4096,4096
AnalogPeak = 32768,32768
Mempak switch =
Rumblepak switch =
C Button D = button(0) button(3)
C Button L = button(4)
Z Trig = button(6)
Start = button(11)
Y Axis = hat(0 Up Down)
DPad U = axis(1-)
C Button U = button(8)
A Button = button(1)
DPad D = axis(1+)
X Axis = hat(0 Left Right)
R Trig = button(7)
DPad R = axis(0+)
DPad L = axis(0-)
C Button R = button(9)
L Trig = button(10)
; 8Bitdo N64 Gamepad_END
```
Why I am submitting this as a bug is because you can see that things clearly don't match in any sense. The most obvious being the A and B buttons. A was set to 1 in the config file (given 0) and B doesn't even exist! It is easy to go in and fix these issues by hand, but there is clearly a mapping error here.The main menu's config input is not correctly passing the values. 

System:
```
pi@retropie:~ $ uname -a
Linux retropie 4.4.34-v7+ #930 SMP Wed Nov 23 15:20:41 GMT 2016 armv7l GNU/Linux
pi@retropie:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 8.0 (jessie)
Release:	8.0
Codename:	jessie
```

joolswills, 2017-04-24:
Not much can be done about some devices, as the IDs map very differently between SDL2 (which we use for ES), and other applications. I assume that's the problem, but I would prefer you started a thread first on the forum, as mentioned when opening an issue as more people are there to advise.

>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

stevenwalton, 2017-05-01:
Sorry, just habitually work here. And already have an account. Just thought I'd mention this since that controller is pretty common. If it isn't going to be fixed feel free to close the thread. Or keep it open as an active bug.

closed 2017-07-02 by joolswills