RetroPie/RetroPie-Setup #1963: gamecondriver incompatible with kernel 4.9.x

joolswills, 2017-04-30:
4.9.24 is now packaged on Raspbian. Gamecondriver is not yet compatible and fails to build.

```
From make.log:
 
DKMS make.log for gamecon_gpio_rpi-1.2 for kernel 4.9.24-v7+ (armv7l)
Sun 30 Apr 04:07:44 UTC 2017
make -C /lib/modules/4.9.24-v7+/build M=/var/lib/dkms/gamecon_gpio_rpi/1.2/build modules
make[1]: Entering directory '/usr/src/linux-headers-4.9.24-v7+'
  CC [M]  /var/lib/dkms/gamecon_gpio_rpi/1.2/build/gamecon_gpio_rpi.o
/var/lib/dkms/gamecon_gpio_rpi/1.2/build/gamecon_gpio_rpi.c:39:27: fatal error: mach/platform.h: No such file or directory
 #include <mach/platform.h>
                           ^
compilation terminated.
scripts/Makefile.build:299: recipe for target '/var/lib/dkms/gamecon_gpio_rpi/1.2/build/gamecon_gpio_rpi.o' failed
make[2]: *** [/var/lib/dkms/gamecon_gpio_rpi/1.2/build/gamecon_gpio_rpi.o] Error 1
Makefile:1490: recipe for target '_module_/var/lib/dkms/gamecon_gpio_rpi/1.2/build' failed
make[1]: *** [_module_/var/lib/dkms/gamecon_gpio_rpi/1.2/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.9.24-v7+'
Makefile:5: recipe for target 'all' failed
make: *** [all] Error 2
```

joao-abrantes, 2017-05-03:
I am facing the same issue :(

turuwhiska, 2017-05-07:
me too any progress?

spfunnell, 2017-05-08:
Yep, same problem here too.  Upgraded RetroPie from 4.1 to 4.2 and DB9 driver no longer works.

joolswills, 2017-05-08:
It's a known confirmed issue. However we don't manage the packages. I will check for updates but we are waiting on the upstream dev

joao-abrantes, 2017-05-16:
Guys, best thing to do is to setup the gamecon & db9 driver before updating the system. if you do this right after a fresh install (following wifi setup, of course), you'll be able to install the driver and have it working. this is not the ideal scenario because you can't update the system, but at least it's working for me!

atarumoroboshi18, 2017-06-05:
Thank you so much @joao-abrantes! I could not for the life of me get the Gamecon drivers working and have poured through tons of forums and posts. Your post was exactly what I needed and my Retropie is working great now!

romain550, 2017-06-13:
I found a temporary solution to update all packages and latest retropie version, without the kernel. You need to start with a clean install of RetroPie
`sudo apt-mark hold raspberrypi*`
To be sure, i disabled all "raspberrypi" package, correct me if it's wrong.
Anyway, that works !

geneworld, 2017-06-30:
Any updates on this working with kernel 4.9x?

joolswills, 2017-07-02:
No.

psyke83, 2017-07-17:
If you want to build a modified package, this should be enough to make it work (but I don't have any controllers to test): 

https://github.com/psyke83/gamecon-gpio-rpi/commit/76ff1220e9ec590bdae2fb5056fa2a9cf66d20da

That's just a hack; the proper way would be to rewrite the driver to find the gpio address using platform_get_resource, like so: https://github.com/raspberrypi/linux/blob/rpi-4.9.y/drivers/char/broadcom/bcm2835-gpiomem.c#L157

psyke83, 2017-07-28:
@joolswills,

I released a test .deb & received a report that the module is working: https://retropie.org.uk/forum/topic/11925/gamecon_gpio_rpi-1-3-kernel-4-9-build-fix

The package was built from extracting the latest build via dpkg-deb and rebuilding. I can't seem to find a proper source repository nor am I sure who to ping in order for upstream to be aware of a possible fix. Can you point me in the right direction? If you want to use my deb as an interim fix, it should work OK at least until the Pi 4 is released ;)

PeterPablo, 2017-07-31:
Thank you @psyke83 for posting your findings. It at least gave me an idea how to proceed with an unrelated driver issue.

marqs85, 2017-08-28:
I've updated "official" driver packages to be compatible with recent kernels. Sources and dkms scripts for [gamecon_gpio_rpi](https://github.com/marqs85/gamecon_gpio_rpi) and [db9_gpio_rpi](https://github.com/marqs85/db9_gpio_rpi) are also now on github.

closed 2017-09-10 by joolswills

knifesk, 2017-12-31:
Same issue.. different kernel version.. 4.14 (and I guess >=4.10) have this same problem

marqs85, 2018-01-03:
I had no issue compiling the module for 4.14.10-v7+ with self-built headers package.

knifesk, 2018-01-03:
I had to re-flash everything... Everything stopped working after I did a
raspi-update and a full apt-get update/upgrade... kernel reported there was
no such module and that installation complained about headers not being
present... after installing the headers (it was already installed and
latest version) it kept complaining about this.. So.. I ended up making a
fresh start... havent updated yet though