RetroPie/RetroPie-Setup #1965: Could not successfully build mupen64plus

gator2003, 2017-05-01:
Getting the following error when installing RetroPie on a fresh 16.04.2 x86_64 machine. The setup script is fresh (07c64e7) from yesterday. Ubuntu is fully updated and rebooted before the basic install is attempted.
```
Could not successfully build mupen64plus - N64 emulator MUPEN64Plus
(mupen64plus-core/projects/unix/libmupen64plus.so.2.0.0 not found).
```

Looking at the logs, it appears the "nasm" package is not installed, or attempting to install as a dependency before the build of mupen64plus starts. The nasm command is missing and so the build fails. Later, when installing lr-mupen64plus, the nasm package is installed.

It appears the nasm package needs to be added as a dependency for mupen64plus.

closed 2017-05-01 by joolswills

INTERZONE13, 2020-04-08:
Hi

Same issue but   "isPlatform "x86" && depends+=(nasm)" is already present in scriptmodules/emulators/mupen64plus.sh" ...

Can you help me please?

Thankss

joolswills, 2020-04-08:
Please open a topic on the RetroPie forum providing more details. Thanks.

commit a3880c22 (2017-05-01): mupen64plus - add nasm dependency for x86 - fixes #1965