RetroPie/RetroPie-Setup #1967: python-dbus missing on fresh Ubuntu install

devildog820, 2017-05-01:
Tried to register a DS4 to 64bit Ubuntu 16.04.2 installed fresh on a MacMini 2011. Kept getting an error that the dbus module was missing. Installing the python-dbus module via apt-get (which also installed python-gi) allowed the script to finish and the controller to register properly.

joolswills, 2017-05-01:
which script ?

devildog820, 2017-05-01:
Ack, sorry about that. Nothing worse than a vague bug report.

Accessing "Bluetooth" from the RetroPie screen in ES. Then choosing "Register and Connect To Bluetooth Device", selecting the DS4, and selecting DisplayYesNo (or any other option on that screen).

Going back into the Bluetooth script caused more python packages to be installed. I included the logs below.

This was the manual "sudo apt-get install python-dbus".
```
Log started: 2017-05-01  14:22:13
Selecting previously unselected package python-dbus.
(Reading database ... 232640 files and directories currently installed.)
Preparing to unpack .../python-dbus_1.2.0-3_amd64.deb ...
Unpacking python-dbus (1.2.0-3) ...
Selecting previously unselected package python-gi.
Preparing to unpack .../python-gi_3.20.0-0ubuntu1_amd64.deb ...
Unpacking python-gi (3.20.0-0ubuntu1) ...
Setting up python-dbus (1.2.0-3) ...
Remove stale byte-compiled files...
Setting up python-gi (3.20.0-0ubuntu1) ...
Log ended: 2017-05-01  14:22:17
```

This was done automatically upon going back into the Bluetooth menu item from the RetroPie menu in ES.
```
Log started: 2017-05-01  14:30:26
Selecting previously unselected package bluetooth.
(Reading database ... 232705 files and directories currently installed.)
Preparing to unpack .../bluetooth_5.37-0ubuntu5_all.deb ...
Unpacking bluetooth (5.37-0ubuntu5) ...
Selecting previously unselected package python-gobject-2.
Preparing to unpack .../python-gobject-2_2.28.6-12ubuntu1_amd64.deb ...
Unpacking python-gobject-2 (2.28.6-12ubuntu1) ...
Selecting previously unselected package python-gobject.
Preparing to unpack .../python-gobject_3.20.0-0ubuntu1_all.deb ...
Unpacking python-gobject (3.20.0-0ubuntu1) ...
Processing triggers for libc-bin (2.23-0ubuntu7) ...
Setting up bluetooth (5.37-0ubuntu5) ...
Setting up python-gobject-2 (2.28.6-12ubuntu1) ...
Setting up python-gobject (3.20.0-0ubuntu1) ...
Processing triggers for libc-bin (2.23-0ubuntu7) ...
Log ended: 2017-05-01  14:30:30

```

joolswills, 2017-05-01:
python-dbus is already included as a dependency but you should not really be using the bluetooth module on x86 anyway - better to use the GUI bluetooth desktop tools.

devildog820, 2017-05-01:
Gotcha. Will do. Not sure why the dbus module didn't install, but I'll use Ubuntu from here on out to register.

joolswills, 2017-05-01:
Although your last two bug reports were valid - please do follow the instructions when opening an issue which clearly states

>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

I prefer the forum to be used as the first point of contact.

closed 2017-05-01 by joolswills