RetroPie/RetroPie-Setup #1978: Running Blazing Star (NEOGEO) on lr-fbalpha (no sound/music on end stage bosses)

Chogin, 2017-05-11:
Pi Model: 3 B

RetroPie Version Used: 4.2.5

USB Devices connected: None

Controller used: Wii U Pro Controller

Error messages received: All my NEOGEO games run flawlessly with lr-fbalpha except this one. When i reach the 1st boss of the game i simply loose the sound and music (with the exception of the insert coin sound). After i defeat the 1st boss and pass to the second stage sound and music are back until i reach the boss at the end of the level and once again exact same issue (no sound or music...). This is bizarre since it only happens 90% of the time. Every now and then i am able to get sound on those boss fights. If i use Mame2003, Mame2010 or lr-fbalpha 2012 i don't have this sound issues but on the other hand all those emulators have a graphical glitch where your ship cannot touch the left side of the screen and this has been fixed on the lr-fbalpha since version v0.2.97.38.

Emulator: lr-fbalpha v0.2.97.40

How to replicate the problem: happens 9 out of 10 times that i reach the boss at stage 1 (it only takes like 1 minute to reach that boss so it is pretty easy to check if this happens to you)

joolswills, 2017-05-11:
>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

>For ideas and enhancements, please also post to the forum first, to see if it is something we would consider and to see if other users would find it useful.

joolswills, 2017-05-11:
If this is a bug in libretro fbalpha - this is still the wrong place for it (it is developed here https://github.com/libretro/fbalpha)

closed 2017-05-11 by joolswills