RetroPie/RetroPie-Setup #1981: DS4 Controller doesn't work in runcommand

mosi0815, 2017-05-14:
Basically it works in none of the menus on the console. 
Button 0 (square) doesn't open the runcommand menu. In menus the left stick (axis 0/1) does from time to time move the cursor for one line but its not usable.
It would be great to get this fixed because DS4 controllers now work out of the box and are one of the best quality full featured controllers on the market.

dankcushions, 2017-05-14:
works for me (wired).

joolswills, 2017-05-14:
Please use the forum for help

>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

closed 2017-05-14 by joolswills