RetroPie/RetroPie-Setup #1982: quake 1 mission packs

symbios24, 2017-05-14:
Hello there, i`m jim

I`m using retropie v4.2 on a raspberry pi2 and installed libretro-tyrquake, quake 1 and episode 5-dopa working fine but mission pack 1 and 2 are not working, when i click to them they return me back to the main menu of retropie i followed the instructions here "https://github.com/RetroPie/RetroPie-Setup/wiki/Quake"

I appreciate if you can help me

ps i cannot post in official retropie forum its very annoying the way they did it.

Thank you

joolswills, 2017-05-14:
What do you mean you cannot post?

symbios24, 2017-05-14:
i dont know how to post i tried some time following instructions from here but it did not work

joolswills, 2017-05-14:
Please be more specific. The forums do work as we have over 9000 user accounts.

symbios24, 2017-05-14:
well not for me anyway thanks for nothing

joolswills, 2017-05-14:
If you are unable to expand on "did not work" then don't expect me to be able to help you with the forum.

closed 2017-05-14 by joolswills

symbios24, 2017-05-14:
useless anyway

joolswills, 2017-05-14:
Well done for making the block list.

ItsNotNitro, 2017-05-25:
thats not good.