RetroPie/RetroPie-Setup #1983: Changes to  mupen64plus.cfg being overwritten.

Twitch0815, 2017-05-16:
Tested on Retropie 4.3
Enabled Hi_Res Texture packs for both Glide and Rice in mupen64plus.cfg
Rice Entry has to be manually entered as it no longer shows by default in the mupen64plus.cfg 
RunCommand is changed to specify Glide and or Rice-HiRes
Upon running either emulatorvia runcommand.sh mupen64plus.cfg variables are reversed.   
This is breaking Hi-Resolution texture for N64

gizmo98, 2017-05-25:
Did you already update RetroPie-Setup script and mupen64plus emulator? I cannot reproduce the problem.

Twitch0815, 2017-05-25:
Yes for example, went to a 3.8 image, where the functionality was working.  Then just updated Rpi setup script and updated mupen64plus from source and the problem replicates itself.

Twitch0815, 2017-05-26:
I also reconfirmed on 4.2, I logged in modified mupen64plus.cfg to enable the Hi-Resolution texture pack.   Specifically # Use high-resolution texture packs if available.
txHiresEnable = False

Then I ran Retropie-Setup updated retropie setup, updated mupen64plus from binary  

Launched a game, the texture pack I loaded did not load, and when going back to mupen 64plus.cfg to check if the variable had stayed as I had just configured it i get 

 cat mupen64plus.cfg | grep txHiresEnabletxHiresEnable = False
pi@retropie:/opt/retropie/configs/n64 $

Here is a time stamp after I just updated mupen64plus.cfg to enable  
-rw-r--r--  1 pi pi  15742 May 26 05:16 mupen64plus.cfg
cat mupen64plus.cfg | grep txHiresEnabletxHiresEnable = True

after I run and exit game with the option for GlideN64HiRes in runcommand

-rw-r--r--  1 pi pi  15743 May 26 05:17 mupen64plus.cfg

and 
recheck variable 
 cat mupen64plus.cfg | grep txHiresEnabletxHiresEnable = False

gizmo98, 2017-05-31:
@Twitch0815 please update setup script and try again.

joolswills, 2017-07-02:
Closing as no reply / assumed working.

closed 2017-07-02 by joolswills

MironV, 2017-11-19:
@joolswills I'm not able to prevent mupen64plus.cfg being rewritten on the latest builds. I've been trying to get a manual controller configuration working and no matter what I do, whenever I start the N64 emulator the CFG file gets overwritten. Is this known?