RetroPie/RetroPie-Setup #1985: framebuffer not usable after exiting mupen64plus / stella

joolswills, 2017-05-20:
May be different causes, but exiting mupen64plus and stella leaves framebuffer in an unusable state. This didn't use to happen with mupen64plus - could be:

* Changes in runcommand where we used to fbset --depth to always reset fb. Will test if that sorts it.
* Changes in SDL2 code.
* Changes in firmware
* Changes in mupen64plus code.

HerbFargus, 2017-05-20:
I know that the audio driver for some reason caused a black screen with certain video plugins with mupen64plus as well but likely unrelated to the framebuffer issue at hand

joolswills, 2017-05-20:
That's unrelated.

Twitch0815, 2017-05-23:
Could this be related to the mupen64plus.cfg having changes overwritten after mupen64plus is launched and closed?

joolswills, 2017-05-28:
I don't think so. It's likely related to the issues in the ticket.

mvgilpatrick, 2017-05-28:
Just FYI on this. I am no developer and don't know linux terribly well, but my ttyecho script at [http://www.humbug.in/2010/utility-to-send-commands-or-data-to-other-terminals-ttypts/](url) allows commands to be sent to the terminal.  I don't know if that means it isn't a frame buffer issue

joolswills, 2017-09-21:
mupen64plus seems ok here when exiting now - will update original issue.

psyke83, 2017-11-16:
On my stretch testing build, I'm unable to reproduce any problem after repeatedly running & closing stella. When you say "unusable state", do you mean a black screen?

For my setup, instead of disabling the fb blank, I still use this commit that I had previously submitted: https://github.com/psyke83/RetroPie-Setup/commit/561036bb510a0f4b9bc8e42167610937fcf4254c

Maybe stella is triggering a framebuffer blank on exit, which is why I don't see the problem? If it's not that, then perhaps SDL 2.0.6 fixed it, or some other update to stretch's packages fixed it.

mediamoshpit, 2018-02-15:
SDL 2.0.6 must have done the job. I've tested this with Jessie and it doesn't seem to be an issue any more.

elilyulko36, 2018-08-06:
What the ... Is this

joolswills, 2019-09-08:
Just retested this. Still an issue here. Screen isn't blank. Just after exiting emulation station no keyboard input seems to work. Not tested against latest Stella code though.

commit b7d651fd (2019-09-11): stella - update to version 6.0.1
 * this also resolves the framebuffer not accepting input once exiting on the rpi - fixes #1985
closed 2019-09-11 by joolswills