RetroPie/RetroPie-Setup #1999: SSH is not usable without a keyboard

AllenKll, 2017-06-04:
In the documentation for setup, https://github.com/RetroPie/RetroPie-Setup/wiki/First-Installation#hardware, It is indicated that SSH can be used instead of a keyboard. However, following the link to setup SSH indicates you need to type commands with a keyboard. https://github.com/retropie/retropie-setup/wiki/SSH#enable-ssh

Seems to be an oversite to no have a simple, "Enable SSH" option somewhere in the menu system, that can be reached with a controller.

joolswills, 2017-06-04:
Please use the forum as indicated when opening an issue.

raspi-config can be launched via RetroPie menu in ES - and can be used with a  joypad.

closed 2017-06-04 by joolswills

AllenKll, 2017-06-04:
Hello,
  I searched the forum, and there is still no answer as to how this can be done.  I believe that maybe it can be done, if so, then this is a documentation bug, if not it is a feature bug.  Either way it is a bug.

AllenKll, 2017-06-04:
Using version 4.2.  Rasberry pi B

dankcushions, 2017-06-04:
https://github.com/RetroPie/RetroPie-Setup/wiki/SSH#enable-ssh

joolswills, 2017-06-04:
It's a support issue. It's not a bug. Missing info in the documentation is not a bug, but as mentioned when opening a ticket, we prefer the forum to be used first. It's as simple as there are more people to advise on things like this there.