RetroPie/RetroPie-Setup #2001: No audio in Mupen64Plus on a non-Pi installation

siddhartha77, 2017-06-04:
PulseAudio is running while EmulationStation is open, and it conflicts with the SDL audio in Mupen64Plus leading to no sound in N64 games. While I thought that pasuspender would solve the problem, it doesn't. Instead, you need to stop PulseAudio from running then start it up again once Mupen64Plus exists.

I have updated [the FAQ](https://github.com/RetroPie/RetroPie-Setup/wiki/RetroPie-on-Debian#no-audio-in-mupen64plus) with the solution.

It requires changes to `/opt/retropie/emulators/mupen64plus/bin/mupen64plus.sh`, which I believe may get written over with any update.

gizmo98, 2017-07-26:
Which operating system/ubuntu version are you using? I had no audio problem with ubuntu 16.04.

joolswills, 2017-09-10:
Closing due to no feedback from OP.

closed 2017-09-10 by joolswills