RetroPie/RetroPie-Setup #2006: libretro-tyrquake (maybe others) cannot launch official Mission Packs.

sduensin, 2017-06-13:
After following the instructions on the wiki to install the official Quake Mission Packs, I discovered that libretro-tyrquake looks in the wrong place when it runs for the main Quake pak files.

For example, it looks for:
`/home/Pi/RetroPie/roms/quake/id1/hipnotic/pak?.pak`
and:
`/home/Pi/RetroPie/roms/quake/id1/id1/pak?.pak`

Obviously `id1/id1` is invalid.  You can fix the issue with an ugly symlink to create an `id1/id1` or you can move the `hipnotic` and `rogue` folders to be parallel to `id1` and update the shell scripts that start the games.

joolswills, 2017-07-02:
What version of RetroPie-Setup script ?

psyke83, 2017-07-25:
@joolswills,

@sduensin is partially correct. It's true that the folder structure is incorrect, but this is the proper layout:

ports/quake/id1
ports/quake/hipnotic
ports/quake/rogue

i.e. the mission pack folders should be parallel and not subfolders of id1. I've sent a PR: https://github.com/RetroPie/RetroPie-Setup/pull/2054

The Wiki will also need to be updated to reflect this change.

FYI (slightly off-topic) - I purchased the Quake pack on Steam in order to test this, but none of the Quake entries launched successfully for me after installing lr-tyrquake (using latest 4.2.12 script) and full version paks, unrelated to the problem with file paths. I tried installing from binary and built from source, but no difference. The runcommand.log always showed "Illegal instruction".

I manually built against this commit (which would have been the latest from when this issue was opened) and now it launches properly: https://github.com/libretro/tyrquake/commit/12d7ca3b04a4d445edd3ef03d303bca6de466134

It's an upstream issue so I'll bisect and try to solve the problem when I have time, but just letting you know in case it causes further confusion.

sduensin, 2017-07-25:
> libretro/tyrquake@12d7ca3

Sorry it's taken so long for me to get back with you on this.  I'm running
4.2.8.

PR #2054 (2017-07-27): lr-tyrquake: correct mission pack paths
hipnotic and rogue folders should be parallel to id1, not subfolders.

Fixes #2006
commit 0d0fc4d4 (2017-07-27): lr-tyrquake: correct mission pack paths

hipnotic and rogue folders should be parallel to id1, not subfolders.

Fixes #2006
closed 2017-07-27 by joolswills