RetroPie/RetroPie-Setup #2021: mupen64plus not installing.

Kilrobot, 2017-06-26:
Im trying to install mupen64plus in retropie but it's giving me an error here is the log : https://pastebin.com/LpYBFq53

What should i do to get mupen64plus to install?

Omnija, 2017-06-29:
I'm experiencing the same issue as well.
[image]

joolswills, 2017-07-01:
Please provide some details of the set-up. Is this on the RPI on Raspbian or ?

Omnija, 2017-07-01:
Retropie img

joolswills, 2017-07-02:
What version of RetroPie-Setup ? (At the top of the window in the RetroPie-Setup gui).

joolswills, 2017-07-02:
Also I don't think these are the same issues. The first post looks like it could be an x86 system rather than a RPI. Version / Set-up details are needed when reporting issues, or I cannot really advise.

I cannot reproduce the problem on either my RPI2 with Raspbian or Ubuntu PC (Ubuntu 17.04). I cannot see what went wrong on your build @Omnija without a log.

Omnija, 2017-07-02:
Heres the log
https://pastebin.com/tz7bvUWc

joolswills, 2017-07-03:
What version of RetroPie-Setup ?

joolswills, 2017-07-03:
@Omnija 

I have reproduced your issue - which is not the same as the original post. It's due to upstream code changes in GLideN64. I will report it there.

joolswills, 2017-07-03:
@Omnija https://github.com/gonetz/GLideN64/pull/1516 - someone already put in a PR. You will have to wait for that to be accepted.

joolswills, 2017-07-03:
@Omnija the referenced PR has been merged. Please try again.

joolswills, 2017-07-03:
@Kilrobot I will require more details of your set-up to advise as mentioned above. @Omnija's issue was unrelated.

Kilrobot, 2017-07-03:
I'm just using mupen64 withouth retropie now and that works

Omnija, 2017-07-03:
@joolswills can't pull retropie update anymore.
https://pastebin.com/Q1LSbbgB

closed 2017-07-04 by joolswills

41joao, 2017-12-08:
Hi,
I'm having the same issue but i'm it on a ubuntuserver 16.04  x86. I tryed on another pc with x64 and it works great but my pc only supports x86 and I can't find any solution. Can somebode please help?!

joolswills, 2017-12-08:
Please use the RetroPie forum for help.