RetroPie/RetroPie-Setup #2026: esthemes.sh remove an installed theme if I try to update it with no connection

meleu, 2017-07-01:
I think the title is clearly explains the issue.

It's related with this piece of code:
https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/supplementary/esthemes.sh#L35-L37

The esthemes removes the current theme before cloning the repo. Therefore, if the cloning procedure fails (e.g.: no connection), the theme was removed.

joolswills, 2017-07-01:
It's not really a bug - but perhaps I can remove the rm workaround now anyway but I will need to test. If someone doesn't have an internet connection they should definitely not try to update anything though, and if it does fail they can retry,

joolswills, 2017-07-01:
(If an internet connection fails, there are a lot of things in RetroPie that could go wrong - eg if some download is half finished etc).

closed 2017-07-01 by joolswills

meleu, 2017-07-01:
I noticed ot when testing my PR for updating all ES themes in an offline system and it removed all the themes.

joolswills, 2017-07-01:
The rm is removed now