RetroPie/RetroPie-Setup #2027: lr-mupen64plus not building

Lobsta888, 2017-07-01:
Installing on Ubuntu 16.04LTS. Have tried a couple of times and keep getting "Could not successfully build lr-mupen64plus - N64 emu - Mupen64Plus + GLideN64 for libretro (/home/michael/RetroPie-Setup/tmp/build/lr-mupen64plus/mupen64plus_libretro.so not found)"

Log : [https://pastebin.com/2WZh8Yvv](url)

joolswills, 2017-07-02:
I don't have a Ubuntu 16.04LTS system handy, but I cannot reproduce the issue on my Ubuntu 17.04 PC. Is your RetroPie-Setup script up to date ? What's the output of `uname -a` (Are you using a 32 bit or 64bit system).

Lobsta888, 2017-07-02:
Yeah, up to date, Was run first from a fresh download, then tried to just remove and reinstall the lr-mupen64 package, then ran the 'Update RetroPie-Setup script. 

uname -a : 

Linux michael-GP62-6QF 4.10.0-27-generic #30~16.04.2-Ubuntu SMP Thu Jun 29 16:07:46 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

joolswills, 2017-07-03:
I cannot reproduce your issue. There was a dependency issue I sorted (which would only happen if installing lr-mupen64plus before anything else). I installed a new Ubuntu 16.04 in Virtualbox, and it compiled fine.

You should make sure your system is up to date - and perhaps remove/reinstall libpng12-dev

I notice your system has a different kernel version from Ubuntu 16.04 - 16.04 uses Kernel 4.8 - perhaps you have some custom / backported packages installed that are causing the issue.

Lobsta888, 2017-07-03:
I'll take a run at that libpng-dev reinstall. 

I have nfi when/why I upgraded the kernel. If I had to guess, it was while trying to get my nvidia graphics drivers to behave. But that ended up being resolved by rolling back the driver to an older version, not backporting anything.

Lobsta888, 2017-07-04:
Removed and reinstalled libpng12-dev. Still a problem
Booted in 4.8.0-57-generic kernel. Still a problem
Exact same issue. 
uname -a > Linux michael-GP62-6QF 4.8.0-57-generic #62~16.04.1-Ubuntu SMP Wed Jun 21 15:58:20 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
log > https://pastebin.com/SrdG6ARP

I tried looking through the git at https://github.com/libretro/mupen64plus-libretro where it clones from and I can't see that file that it can't find (mupen64plus_libretro.so). Is that something that should be possible? Or is that a file that is supposed to be build and only briefly exists while installing? Is there some other file from the git that isn't being cloned right? I've tried cleaning the source folder and installing from source again and it makes no difference. 

I'm not sure if it's related, but the N64 emulator doesn't work in emulation station. The default emulator will load a rom, and I can hear the audio, and I can tell from that that the controller is working and the game is loaded (I can start the first level in Goldeneye for example). But no video ever shows. Same results in a couple of roms. Changing to any of the other listed emulators in the runcommand menu instead just causes some repeating screen changing and no game ever loads. But when I installed Mupen64Plus-Qt, all roms load and play fine. Bloody frustrating. All other emulators I've tried (SNES, PSX) all work fine through emulationstation. 

Any more ideas? I really don't want to do a fresh OS install!

dankcushions, 2017-07-04:
The .so file is the compiled binary.  The setup script checks for its presence at the end to see if the build has been successful or not.

your build issue appears to be a GLideN64 one:
```
./GLideN64/src/GLideNHQ/TxImage.o: In function `TxImage::getPNGInfo(_IO_FILE*, png_struct_def**, png_info_def**)':
TxImage.cpp:(.text+0xcc): undefined reference to `png_set_longjmp_fn'
./GLideN64/src/GLideNHQ/TxImage.o: In function `TxImage::writePNG(unsigned char*, _IO_FILE*, int, int, int, unsigned short, unsigned char*)':
TxImage.cpp:(.text+0x4e3): undefined reference to `png_set_longjmp_fn'
collect2: error: ld returned 1 exit status
Makefile:304: recipe for target 'mupen64plus_libretro.so' failed
make: *** [mupen64plus_libretro.so] Error 1
Removing additional swap
```
I think the GLideN64 version in is static so not sure what could be up.

joolswills, 2017-07-04:
As I can't reproduce the issue on a fresh 16.04lts, I can only suggest you try a new install (or in virtualbox). Perhaps you have a newer backport of libpng-dev installed that is incompatible or something.

joolswills, 2017-07-04:
Or an older libpng-dev ? Is your system completely up to date ?

what's the output of `apt-cache showpkg libpng-dev` ?

joolswills, 2017-07-04:
edit sorry that should be `apt-cache showpkg libpng12-dev`

joolswills, 2017-07-04:
you can also try `sudo apt-get install libpng16-dev` although this was not needed on my system. If you already have this installed perhaps it's confusing the build - try removing libpng16-dev in that case.

Lobsta888, 2017-07-05:
apt-cache showpkg libpng12-dev
Package: libpng12-dev
Versions: 
1.2.54-1ubuntu1 (/var/lib/apt/lists/au.archive.ubuntu.com_ubuntu_dists_xenial_main_binary-amd64_Packages) (/var/lib/dpkg/status)
 Description Language: 
                 File: /var/lib/apt/lists/au.archive.ubuntu.com_ubuntu_dists_xenial_main_binary-amd64_Packages
                  MD5: 748198d86eeddc4aa1418e7d1311203c
 Description Language: 
                 File: /var/lib/apt/lists/au.archive.ubuntu.com_ubuntu_dists_xenial_main_binary-i386_Packages
                  MD5: 748198d86eeddc4aa1418e7d1311203c
 Description Language: en_AU
                 File: /var/lib/apt/lists/au.archive.ubuntu.com_ubuntu_dists_xenial_main_i18n_Translation-en%5fAU
                  MD5: 748198d86eeddc4aa1418e7d1311203c
 Description Language: en
                 File: /var/lib/apt/lists/au.archive.ubuntu.com_ubuntu_dists_xenial_main_i18n_Translation-en
                  MD5: 748198d86eeddc4aa1418e7d1311203c

Reverse Depends: 
  libpng12-0,libpng12-dev 1.2.8rel-7
  libreoffice-subsequentcheckbase,libpng12-dev
  libgdk-pixbuf2.0-dev,libpng12-dev
  libpng16-tools:i386,libpng12-dev
  libpng16-tools:i386,libpng12-dev
  libpng16-devtools:i386,libpng12-dev
  libpng16-devtools:i386,libpng12-dev
  libpng16-dev:i386,libpng12-dev
  libpng16-dev:i386,libpng12-dev
  libvigraimpex-dev,libpng12-dev
  libreoffice-subsequentcheckbase,libpng12-dev
  libpolybori-dev,libpng12-dev
  libpng16-tools,libpng12-dev
  libpng16-tools,libpng12-dev
  libpng16-devtools,libpng12-dev
  libpng16-devtools,libpng12-dev
  libpng16-dev,libpng12-dev
  libpng16-dev,libpng12-dev
  libcamlimages-ocaml-dev,libpng12-dev
  libalien-sdl-dev-perl,libpng12-dev
  games-c++-dev,libpng12-dev
  libpng12-dev:i386,libpng12-dev 1.2.54-1ubuntu1
  libpng12-dev:i386,libpng12-dev 1.2.54-1ubuntu1
  libpng12-0:i386,libpng12-dev 1.2.8rel-7
  libpng12-0:i386,libpng12-dev 1.2.8rel-7
  libpng12-0,libpng12-dev 1.2.8rel-7
  libnux-4.0-dev,libpng12-dev 1.2.44
  libgdk-pixbuf2.0-dev,libpng12-dev
Dependencies: 
1.2.54-1ubuntu1 - libpng12-0 (5 1.2.54-1ubuntu1) zlib1g-dev (0 (null)) libpng-dev (0 (null)) libpng12-0-dev (0 (null)) libpng2 (3 1.0.12-3) libpng2-dev (0 (null)) libpng12-0-dev (0 (null)) libpng3-dev (1 1.2.5) libpng-dev:i386 (0 (null)) libpng12-0-dev:i386 (0 (null)) libpng12-0-dev:i386 (0 (null)) libpng3-dev:i386 (1 1.2.5) libpng12-dev:i386 (35 1.2.54-1ubuntu1) libpng12-dev:i386 (38 1.2.54-1ubuntu1) libpng2-dev:i386 (0 (null)) 
Provides: 
1.2.54-1ubuntu1 - libpng3-dev (= ) libpng12-0-dev (= ) libpng-dev (= ) 
Reverse Provides: 

Installing libpng16-dev removed libpng12-dev. Then trying to install lr-mupen64plus again removed 16 and reinstalled 12. And also didn't work... 

This is killing me!

joolswills, 2017-07-05:
sorry but I cannot advise further as I cannot reproduce the issue on my 16.04 set-up.

Lobsta888, 2017-07-05:
All good. I'm just resigned to having to play n64 separate to retropie for now. Might fresh install 16.04 if I get super motivated one day. Thanks for the help anyway.

closed 2017-07-05 by Lobsta888

yoisel, 2018-03-04:
I am also facing this problem on xubuntu 16.04, tried first on a 32 bit system, then on x64, then on a virtualbox with a clean install, lr-mupen64plus still does not compile.

joolswills, 2018-03-04:
This ticket is closed. Please use the forum first and if we can confirm an issue we can open a ticket.