RetroPie/RetroPie-Setup #2032: added support for colecovision in libretro blue-msx

bigboo3000, 2017-07-06:
Hi
FIY colecovision support have been added in libretro:
https://github.com/libretro/blueMSX-libretro/pull/41#event-1148586688

joolswills, 2017-07-07:
Thanks. You can install from source to get the latest code. Binaries will be updated soon.

joolswills, 2017-07-07:
(I will look into adding the emulator as an optional emulator for coleco also - but it needs some testing first). Happy for anyone to test and report back (the forum would be the best place for this).

joolswills, 2017-07-07:
I gave it a quick test. Unfortunately it doesn't automatically switch to colecovision when loading a colecovision rom - you need to manually change the core options.

Because we share the core options between all systems, this is a problem as we can't set it to colecovision without it defaulting to that if loading MSX games. Ideally it would autodetect - this would need to be changed in the libretro code.

joolswills, 2017-07-07:
If we had core options per system this could work around it. I have thought about doing this before, but it would mean old core options would be lost if we switched. Something I would want to discuss on the forum.

joolswills, 2017-07-07:
Should be possible to force a different core options just for this system - which I will test. Not ideal, but should make it work.

commit 0bffaabb (2017-07-12): lr-bluemsx - added colecovision support - fixes #2032
 * document msx/colecovision bios / rom locations
 * added to coleco system
 * use a separate retroarch-core-options.cfg so as to force the colecovision system
commit 60b03ffc (2017-07-12): Merge pull request #2034 from joolswills/bluemsx_coleco

lr-bluemsx - added colecovision support - fixes #2032
closed 2017-07-12 by joolswills