RetroPie/RetroPie-Setup #2052: [RPI3 ; pcsx-rearmed core (non lr !)] builtin_gpu enhanced resolution causes segmentation fault

Aryetis, 2017-07-22:
Hello,

Well I think everything is said within the title. Whenever I try to turn on "Enhanced resolution (slow)" option from the builtin_gpu plugin the emulator crashes back to Retropie with a segmentation fault error. Depending of the games I tried I can sometimes watch the cutscences, navigate within the main menu, get some very basic 3D elements (eg: car preview from Colin McRae Rally 2.0). But as soon as the real game begins (eg: a race in Colin McRae Rally 2.0) the emulator crashes everytime, at the very same point.

Am I the only one in this situation ?

I already reported the bug on pcsx-rearmed's github. Notaz couldn't reproduce the bug so we assumed it was not a upstream bug but due to some retropie shenanigans. ( link to pcsx-rearmed related ticket : https://github.com/notaz/pcsx_rearmed/issues/89 )

Thanks for reading, and even more to those answering.
--------------------------------------------------------------------
Specifications : Raspberry Pi 3 , Retropie 4.2.10
pcsx version : binaries and built from source using Retropie_setup.sh (same results)
pcsx.cfg file : https://pastebin.com/q35icuwZ
games tested : Crash Bandicoot 3, Driver, Colin McRae Rally 2.0, Colony Wars, etc

gizmo98, 2017-07-26:
We use notaz repo. There is no retropie fork. Is there a reason you prefer standalone? The only reason to use pcsx-rearmed in the past was better performance on a pi1/Zero.

Aryetis, 2017-07-26:
I wanted to use pcsx-rearmed because it seems to have better performance on a handful of titles (eg: "colin mcrae rally 2.0") I get some dropped frames and sometimes choppy audio using lr-pcsx-rearmed whereas if I remember correctly it was running all smoothly on pcsx-rearmed (couple months ago when "enhanced resolution" wasn't breaking the emulation).
Yes it's pretty nitpicky, but anyways, the lack of analog support in pcsx-rearmed core is kinda of a deal breaker.

joolswills, 2017-09-17:
Unable to reproduce it here (Tested with rtype delta). Make sure you are not overclocked - please use the forum for further help though rather than here - if others can confirm the issue we can reopen the ticket.

closed 2017-09-17 by joolswills