RetroPie/RetroPie-Setup #2055: lr-tyrquake "13284 Illegal instruction" Error - confirmed with official Pak's

Dochartaigh, 2017-07-26:
Per this topic: [https://www.reddit.com/r/RetroPie/comments/6pjyh7/lrtyrquake_not_loading/](https://www.reddit.com/r/RetroPie/comments/6pjyh7/lrtyrquake_not_loading/)

At least 2 people have with identical error logs when running official (NON-Mission) Quake pak's in id1 directory, with lr-tyrquake. Retropie version 4.2.10. Logs from Dev/shm:

`Parameters: Executing: /opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-tyrquake/tyrquake_libretro.so --config /opt/retropie/configs/ports/quake/retroarch.cfg "/home/pi/RetroPie/roms/ports/quake/id1/pak0.pak" --appendconfig /dev/shm/retroarch.cfg /opt/retropie/supplementary/runcommand/runcommand.sh: line 1005: 13284 Illegal instruction /opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-tyrquake/tyrquake_libretro.so --config /opt/retropie/configs/ports/quake/retroarch.cfg "/home/pi/RetroPie/roms/ports/quake/id1/pak0.pak" --appendconfig /dev/shm/retroarch.cfg`

`quake lr-tyrquake /home/pi/RetroPie/roms/ports/quake/id1/pak0.pak /opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-tyrquake/tyrquake_libretro.so --config /opt/retropie/configs/ports/quake/retroarch.cfg "/home/pi/RetroPie/roms/ports/quake/id1/pak0.pak"`

Dagnillo, 2017-07-26:
I'm the other person.... Running 4.2 with default HDMI settings.  My log showed the same error, regardless of what PAK files were present.

psyke83, 2017-07-26:
Relevant (I haven't yet had time to bisect the exact commit): https://github.com/RetroPie/RetroPie-Setup/issues/2006#issuecomment-317601846

psyke83, 2017-07-27:
I've bisected the problem to this commit: https://github.com/libretro/tyrquake/commit/2b0c9f232a9505862f10abc45558c0703a9d30b8

Latest master works is also working fine if you revert just that commit. Any ideas on why it's producing illegal instructions?

Dochartaigh, 2017-07-27:
Hi psyke83, thank you for the comments but to be completely honest I don't even know what a "commit" is in reference to RetroPie...or how to revert just that commit to make Quake work. Is this some line of code I could change in a certain file and it will make Quake work again? Or is it simply showing that a change was made, it bricked Quake, and you've narrowed it down to this certain commit?

psyke83, 2017-07-27:
Not to worry - that info is more relevant to @joolswills to help diagnose the issue. The problem is likely to need fixing upstream in the lr-tyquake repository.

Dochartaigh, 2017-07-27:
Thanks. I will await the fix!

Dagnillo, 2017-07-27:
Okay so I uninstalled then reinstalled from binary, now it crashes but the error has changed to this:

line 1005: 6044 illegal instruction

sxales, 2017-07-31:
I'll be number 3, I am made the mistake of updating today so I am getting the same error.

`Parameters: 
Executing: /opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-tyrquake/tyrquake_libretro.so --config /opt/retropie/configs/ports/quake/retroarch.cfg "/home/osmc/RetroPie/roms/ports/quake/id1/pak0.pak" --appendconfig /dev/shm/retroarch.cfg
/opt/retropie/supplementary/runcommand/runcommand.sh: line 1005:  1693 Illegal instruction     /opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-tyrquake/tyrquake_libretro.so --config /opt/retropie/configs/ports/quake/retroarch.cfg "/home/osmc/RetroPie/roms/ports/quake/id1/pak0.pak" --appendconfig /dev/shm/retroarch.cfg
`

anunna777, 2017-08-01:
Yep same here. I updated everything the other day... only Quake wouldn't launch. Re-installed lr-tyrquake but it didn't help. Uninstalled it then re-installed from source... nothing again. I've changed to different wads, looked at different launch scripts "Launching Quake and it's extras (1 year anniversary release)" and nothing works... but it did a few days ago.

-this log is with cyperghosts launch script (and folder layout for mission packs) but the same happens with official launch script on the wiki - just with a different "Illegal instruction" number. 

Parameters: 
Executing: /opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-tyrquake/tyrquake_libretro.so --config /opt/retropie/configs/ports/quake/retroarch.cfg "/home/pi/RetroPie/roms/ports/quake/id1/pak0.pak" --appendconfig /dev/shm/retroarch.cfg
/opt/retropie/supplementary/runcommand/runcommand.sh: line 1005: 19832 Illegal instruction     /opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-tyrquake/tyrquake_libretro.so --config /opt/retropie/configs/ports/quake/retroarch.cfg "/home/pi/RetroPie/roms/ports/quake/id1/pak0.pak" --appendconfig /dev/shm/retroarch.cfg

(I'm new to posting anything anywhere but want this sorted so thought I'd share my log).

joolswills, 2017-08-01:
Thanks. We don't need any more logs - it's a known issue and needs to be reported/fixed upstream.

Dochartaigh, 2017-08-01:
Joolswills, should we report it upstream? If so, how do we do that?

psyke83, 2017-08-02:
@joolswills Here it is, would appreciate feedback if it doesn't look right: https://github.com/libretro/tyrquake/pull/47

@Dochartaigh. No, it's already done. Just wait for the fix to be merged - thanks.

joolswills, 2017-08-02:
Thanks. Will rebuild bins later today.

joolswills, 2017-08-05:
This is fixed.

closed 2017-08-05 by joolswills