RetroPie/RetroPie-Setup #2064: Failed saving config to /opt/retropie/configs/psx/retroarch.cfg

luizoti, 2017-07-30:
Good evening,

I think this is a bug, I'm using retropie 4.2.13 and I get this error when saving the settings in the `Settings >>> Save Current Configuration` menu.

Deleting the file `/opt/retropie/configs/psx/retroarch.cfg`
The retroarch can save normally, but I noticed that when using the` Save Core Overrides` option the problem happens again.

This is normal?

joolswills, 2017-07-30:
You need to use the forum as detailed when opening an issue.

closed 2017-07-30 by joolswills

joolswills, 2017-07-30:
>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.