RetroPie/RetroPie-Setup #2075: emulationstation build fails on Ubuntu 16.04 LTS

chiefy, 2017-08-08:
```
Log started at: Mon Aug  7 20:40:23 EDT 2017

RetroPie-Setup version: 4.2.14 (0449ee0)
System: Linux retropie 4.10.0-30-generic #34~16.04.1-Ubuntu SMP Wed Aug 2 02:13:56 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

= = = = = = = = = = = = = = = = = = = = =
Installing dependencies for 'emulationstation' : EmulationStation - Frontend used by RetroPie for launching emulators
= = = = = = = = = = = = = = = = = = = = =

~/RetroPie-Setup/tmp/build/emulationstation ~/RetroPie-Setup

= = = = = = = = = = = = = = = = = = = = =
Getting sources for 'emulationstation' : EmulationStation - Frontend used by RetroPie for launching emulators
= = = = = = = = = = = = = = = = = = = = =

git clone --recursive --depth 1 --branch v2.4.1 "https://github.com/RetroPie/EmulationStation" "/home/chief/RetroPie-Setup/tmp/build/emulationstation"
Cloning into '/home/chief/RetroPie-Setup/tmp/build/emulationstation'...
Note: checking out 'b28fa2ce9f6a4f9b0d8ad3af7bf09fc008c80d59'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

Submodule 'external/pugixml' (https://github.com/zeux/pugixml.git) registered for path 'external/pugixml'
Cloning into 'external/pugixml'...
Submodule path 'external/pugixml': checked out 'd2deb420bc70369faa12785df2b5dd4d390e523d'
~/RetroPie-Setup
~/RetroPie-Setup/tmp/build/emulationstation ~/RetroPie-Setup

= = = = = = = = = = = = = = = = = = = = =
Building 'emulationstation' : EmulationStation - Frontend used by RetroPie for launching emulators
= = = = = = = = = = = = = = = = = = = = =

-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
Looking for bcm_host.h
bcm_host.h not found
Looking for libMali.so
libMali.so not found
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so  
-- Found Freetype: /usr/lib/x86_64-linux-gnu/libfreetype.so  
-- Looking for FreeImage...
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'freeimage'
--   No package 'freeimage' found
-- Found FreeImage: optimized;/usr/lib/x86_64-linux-gnu/libfreeimage.so;debug;/usr/lib/x86_64-linux-gnu/libfreeimage.so
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found SDL2: /usr/lib/x86_64-linux-gnu/libSDL2main.a;/usr/lib/x86_64-linux-gnu/libSDL2.so;-lpthread  
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   filesystem
--   date_time
--   locale
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "2.91.0") 
-- Found CURL: /usr/lib/x86_64-linux-gnu/libcurl.so (found version "7.47.0") 
-- Checking for module 'libvlc>=1.0.0'
--   Found libvlc, version 2.2.2
-- VLC library found
-- Found VLC: /usr/include/vlc  
-- Found ALSA: /usr/lib/x86_64-linux-gnu/libasound.so (found version "1.1.0") 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/chief/RetroPie-Setup/tmp/build/emulationstation
Scanning dependencies of target pugixml
[  1%] Building CXX object external/pugixml/CMakeFiles/pugixml.dir/src/pugixml.cpp.o
Scanning dependencies of target nanosvg
[  1%] Building CXX object external/nanosvg/CMakeFiles/nanosvg.dir/src/nanosvg_impl.cpp.o
[  2%] Linking CXX static library ../../libnanosvg.a
[  2%] Built target nanosvg
[  3%] Linking CXX static library ../../libpugixml.a
[  3%] Built target pugixml
Scanning dependencies of target es-core
[  4%] Building CXX object es-core/CMakeFiles/es-core.dir/src/AudioManager.cpp.o
[  5%] Building CXX object es-core/CMakeFiles/es-core.dir/src/GuiComponent.cpp.o
[  6%] Building CXX object es-core/CMakeFiles/es-core.dir/src/HelpStyle.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
es-core/CMakeFiles/es-core.dir/build.make:110: recipe for target 'es-core/CMakeFiles/es-core.dir/src/HelpStyle.cpp.o' failed
make[2]: *** [es-core/CMakeFiles/es-core.dir/src/HelpStyle.cpp.o] Error 4
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:217: recipe for target 'es-core/CMakeFiles/es-core.dir/all' failed
make[1]: *** [es-core/CMakeFiles/es-core.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
~/RetroPie-Setup
Could not successfully build emulationstation - EmulationStation - Frontend used by RetroPie for launching emulators (/home/chief/RetroPie-Setup/tmp/build/emulationstation/emulationstation not found).

Log ended at: Mon Aug  7 20:40:44 EDT 2017
Total running time: 0 hours, 0 mins, 21 secs
```

biopendent, 2017-08-08:
Try downloading it as a zip to home folder extract and compile?

chiefy, 2017-08-08:
@biopendent don't even have to do that, as I stated: I can compile/install just fine doing it manually, for whatever reason using the setup script causes it to barf on `es-core/CMakeFiles/es-core.dir/src/HelpStyle.cpp.o`

joolswills, 2017-08-08:
Looks like you ran out of memory - But please use the forum as detailed when opening an issue

>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

>For ideas and enhancements, please also post to the forum first, to see if it is something we would consider and to see if other users would find it useful.

closed 2017-08-08 by joolswills

chiefy, 2017-08-09:
@joolswills yeah - you're right. I re-created my VM w/ 2GB of memory and it worked fine. Thanks.