RetroPie/RetroPie-Setup #2078: Add native support for Joy-Cons (using them as pair)

eRKoo32, 2017-08-09:
Its already possible with https://github.com/jgeumlek/MoltenGamepad and https://github.com/riking/joycon , but they do not work very well with retropie.

cainkirisgai, 2017-12-12:
i too would also like this. im sure there are more then just us as well, i did try both of those but for some reason it either doesnt compile correctly or it will not detect the joycons although they are found when i check the connected devices

frntz, 2018-02-04:
@eRKoo32, @cainkirisgai Do you have any updates about it guys ? I'm thinking about building a portable console by mixing this two projects https://timee1994.weebly.com/nintimdo-rp.html, https://www.thingiverse.com/thing:2173035 but I'm stuck until I could find a way to make the 2 joy cons act as 1 joypad ...

eRKoo32, 2018-02-04:
I figured it out back in the August, but sadly, I don't remember how. I sold my Switch few days after that.
I think developer of

frntz, 2018-02-05:
@eRKoo32 thanks for your answer, now I know that it's possible ! I just have to dig it more :)

eRKoo32, 2018-02-05:
I'm really sorry I can't help you more, my memory is pretty bad. But I think I contacted dev of one of the 2 links I posted above (think it was Riking)... and he messaged me back few months later that he implemented that feature. Sadly I don't have that twitter account anymore.

Teeqla, 2018-03-14:
Did you figure it out? 
Stuck on the same problem..

Dogy007, 2018-06-25:
I also have this issue

theofficialgman, 2021-06-20:
this issue should be able to be closed
joycons can be used as a pair by using joycond (as is done on nintendo switch linux) https://github.com/DanielOgorchock/joycond
paired with a kernel which has the HID-Nintendo driver module (https://github.com/nicman23/dkms-hid-nintendo)

cmitu, 2022-06-15:
As mentioned above, the controllers should be supported with [joycond](https://github.com/DanielOgorchock/joycond).
Install instructions for Joycon support is covered in https://retropie.org.uk/docs/Nintendo-Switch-Controllers/.

closed 2022-06-15 by cmitu