RetroPie/RetroPie-Setup #2087: Retropie ... configuration instead of 13 games.

erisosa81, 2017-08-14:
I just downloaded retropie using my MAC and Raspberry Pi 3 wireless, but for some reason it does not say 13 games like everyone else it just says configuration, I tried to connect to my wireless home network using the credential transfer and it has been unsuccessful can anyone help me?

dankcushions, 2017-08-14:
> I just downloaded retropie using my MAC and Raspberry Pi 3 wireless, but for some reason it does not say 13 games like everyone else it just says configuration

this is normal. perhaps you're looking at screenshots from older versions?

> I tried to connect to my wireless home network using the credential transfer and it has been unsuccessful can anyone help me?

please use the forum for support: retropie.org.uk/forum

robertybob, 2017-08-14:
@erisosa81 This confused me too. I hadn't started from scratch since RP 3.8 so I wondered why the Ports, Dreamcast etc. menus were missing!

closed 2017-08-14 by joolswills