RetroPie/RetroPie-Setup #2095: Problem Font.cpp

lucasmarsola, 2017-08-23:
I think this is a bug. When I open emulationstation, press any button, this happens:
`emulationstation: /home/marsola/RetroPie-Setup/tmp/build/emulationstation/es-core/src/resources/Font.cpp:129: static UnicodeChar Font::readUnicodeChar(const string&, std::size_t&): Assertion `(c & 0xC0) != 0x80' failed. `

drwhoelse, 2017-08-24:
I have similar issue.  This happened right after updating Retropie-Setup 4.2.18 and then updating everything within Retropie on my Intel i7 PC with Xubuntu 17.04 64-bit.

I have different result and error message.  ES crashes right after selecting any one of consoles, and then the screen stays black for a while (not showing gamelist yet).  The "es_log.txt" file doesn't have the error message for this.  Error is visible when using in terminal with emulationstation command... It reads:

emulationstation: /home/ubu/RetroPie-Setup/tmp/build/emulationstation/es-core/src/resources/Font.cpp:129: static UnicodeChar Font::readUnicodeChar(const string&, std::size_t&): Assertion `(c & 0xC0) != 0x80' failed.
Aborted (core dumped)

By the way, Xubuntu 16.04.3 LTS 32-bit on my PC's second partition does not produce this error.

EricFelbs, 2017-08-25:
I'm getting the same error it crashes every time I try to select an emulator

falmear, 2017-08-27:
Starting program: /opt/retropie/supplementary/emulationstation/emulationstation 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffe5a2f700 (LWP 16968)]
emulationstation: /home/retropie/RetroPie-Setup/tmp/build/emulationstation/es-core/src/resources/Font.cpp:132: static UnicodeChar Font::readUnicodeChar(const string&, std::size_t&): Assertion `false' failed.

Thread 1 "emulationstatio" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:58
58      ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:58
#1  0x00007ffff55bf37a in __GI_abort () at abort.c:89
#2  0x00007ffff55b5b47 in __assert_fail_base (fmt=<optimized out>, assertion=assertion@entry=0x555555a9c389 "false", file=file@entry=0x555555ab3e20 "/home/retropie/RetroPie-Setup/tmp/build/emulationstation/es-core/src/resources/Font.cpp", line=line@entry=132, 
    function=function@entry=0x555555ab4580 <Font::readUnicodeChar(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned long&)::__PRETTY_FUNCTION__> "static UnicodeChar Font::readUnicodeChar(const string&, std::size_t&)")
    at assert.c:92
#3  0x00007ffff55b5bf2 in __GI___assert_fail (assertion=0x555555a9c389 "false", file=0x555555ab3e20 "/home/retropie/RetroPie-Setup/tmp/build/emulationstation/es-core/src/resources/Font.cpp", line=132, 
    function=0x555555ab4580 <Font::readUnicodeChar(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned long&)::__PRETTY_FUNCTION__> "static UnicodeChar Font::readUnicodeChar(const string&, std::size_t&)") at assert.c:101
#4  0x00005555558fa641 in Font::readUnicodeChar(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned long&) ()
#5  0x00005555558fd7b5 in Font::sizeText(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, float) ()
#6  0x00005555558e54c2 in TextComponent::calculateExtent() ()
#7  0x00005555558e570c in TextComponent::onTextChanged() ()
#8  0x00005555558e6421 in TextComponent::TextComponent(Window*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::shared_ptr<Font> const&, unsigned int, Alignment, Eigen::Matrix<float, 3, 1, 0, 3, 1>, Eigen::Matrix<float, 2, 1, 0, 2, 1>, unsigned int) ()
#9  0x00005555558d8d7d in HelpComponent::updateGrid() ()
#10 0x00005555558c322d in Window::setHelpPrompts(std::vector<std::pair<char const*, char const*>, std::allocator<std::pair<char const*, char const*> > > const&, HelpStyle const&) ()
#11 0x000055555588d772 in GuiComponent::updateHelpPrompts() ()
#12 0x0000555555866310 in std::_Function_handler<void (), ViewController::playViewTransition()::{lambda()#2}>::_M_invoke(std::_Any_data const&) ()
#13 0x00005555558c4a0c in AnimationController::~AnimationController() ()
#14 0x00005555558c4a49 in AnimationController::~AnimationController() ()
#15 0x000055555588d108 in GuiComponent::advanceAnimation(unsigned char, unsigned int) ()
#16 0x000055555588d15a in GuiComponent::updateSelf(int) ()
#17 0x00005555558c230b in Window::update(int) ()
#18 0x0000555555784f4d in main ()

Gyges-of-lydia, 2017-08-28:
+1 
(ubuntu 17.04)
fresh install.

joolswills, 2017-08-29:
this should be opened here - https://github.com/RetroPie/EmulationStation/ but see https://github.com/RetroPie/EmulationStation/pull/226

pjft, 2017-08-29:
Sorry for that.

If someone could test whether this PR effectively addresses it in Ubuntu, that would be extremely helpful as I have no access to an Ubuntu setup.

https://github.com/RetroPie/EmulationStation/pull/231

Thanks for flagging this, and apologies for the trouble.

Gyges-of-lydia, 2017-08-29:
It Helped! i Updated from the setup script and it worked like charm! Many Many Thanks!

drwhoelse, 2017-08-30:
> RetroPie/EmulationStation#231

I can confirm that ES V2.6.1RP is working properly on Xubuntu 17.04.
Doesn’t crash since then.  Happy camper here!

joolswills, 2017-08-31:
This is resolved now.

closed 2017-08-31 by joolswills

gaenserich, 2017-09-28:
I still get this error on ES 2.6.3rp on RetroPie 4.3 (RPi3). I've reinstalled and rebuilt from source several times to try to fix this, but still get this crash and error when browsing to the megadrive or pc pages:
```sh
emulationstation: /home/pi/RetroPie-Setup/tmp/build/emulationstation/es-core/src/resources/Font.cpp:129: static UnicodeChar Font::readUnicodeChar(const string&, std::size_t&): Assertion `(c & 0xC0) != 0x80' failed.
Aborted
```
If I remove the megadrive or pc rom directories, the crash and error still come about when browsing to a different system page which previously hadn't caused the crash. Adding another system (ZX Spectrum) does not change this, either (that is, the crash and error still occur when attempting to open Megadrive or PC pages).

-I use the Selph Scraper with video thumbnail support, with the `images` directory and `gamelist.xml` file directly in the rom folders.

Other System Info:
`uname -a`: `Linux retropie 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux`

`dpkg-query -W -f='${binary:Package} ${Version}\n'`:
```sh
acl 2.2.52-2
acpi-support-base 0.142-6
acpid 1:2.0.23-2
adduser 3.113+nmu3
advancecomp 1.19-1
alsa-utils 1.0.28-1
apt 1.0.9.8.4
apt-listchanges 2.85.13+nmu1
apt-transport-https 1.0.9.8.4
apt-utils 1.0.9.8.4
aptitude 0.6.11-1
aptitude-common 0.6.11-1
autoconf 2.69-8
automake 1:1.14.1-4+deb8u1
autopoint 0.19.3-2
autotools-dev 20140911.1
avahi-daemon 0.6.31-5
base-files 8+rpi1
base-passwd 3.5.37
bash 4.3-11+deb8u1
bash-completion 1:2.1-4
bind9-host 1:9.9.5.dfsg-9+deb8u13
binutils 2.25-5+deb8u1
bison 2:3.0.2.dfsg-2
bluetooth 5.23-2+rpi2
bluez 5.23-2+rpi2
bluez-firmware 1.2-3+rpi2
bsdmainutils 9.0.6
bsdutils 1:2.25.2-6
build-essential 11.7
bzip2 1.0.6-7
ca-certificates 20141019+deb8u3
cifs-utils 2:6.4-1
cmake 3.6.2-2~bpo8+1
cmake-data 3.6.2-2~bpo8+1
console-setup 1.123
console-setup-linux 1.123
coreutils 8.23-4
cpio 2.11+dfsg-4.1+deb8u1
cpp 4:4.9.2-2
cpp-4.9 4.9.2-10
cpufreqd 2.4.2-2
cpufrequtils 008-1
crda 3.13-1
cron 3.0pl1-127+deb8u1
curl 7.38.0-4+deb8u5
dash 0.5.7-4
dbus 1.8.22-0+deb8u1
dc 1.06.95-9
debconf 1.5.56+deb8u1
debconf-i18n 1.5.56+deb8u1
debconf-utils 1.5.56+deb8u1
debhelper 10.2.2~bpo8+1
debianutils 4.4
device-tree-compiler 1.4.1-1+rpi1
devscripts 2.15.3+deb8u1
[... 713 lines cut ...]
whois 5.2.7
wireless-regdb 2016.06.10-1~deb8u1
wireless-tools 30~pre9-8
wpasupplicant 2.3-1+deb8u4
x11-common 1:7.7+16
x11proto-core-dev 7.0.31-1
x11proto-damage-dev 1:1.2.1-2
x11proto-dri2-dev 2.8-2
x11proto-fixes-dev 1:5.0-2
x11proto-gl-dev 1.4.17-1
x11proto-input-dev 2.3.1-1
x11proto-kb-dev 1.0.6-2
x11proto-xext-dev 7.3.0-1
x11proto-xf86vidmode-dev 2.3.1-2
xauth 1:1.0.9-1
xdg-user-dirs 0.15-2
xkb-data 2.12-1
xml-core 0.13+nmu2
xmlstarlet 1.6.1-1
xorg-sgml-doctools 1:1.11-1
xtrans-dev 1.3.5-1
xz-utils 5.1.1alpha+20120614-2
zlib1g:armhf 1:1.2.8.dfsg-2
zlib1g-dev:armhf 1:1.2.8.dfsg-2
```

pjft, 2017-09-29:
@gaenserich we might want to track this in a separate bug then.

Just to be clear, though, you're getting this error on a _Pi_? I haven't seen any such reports prior to this on a Pi, hence the question, as it might be slightly unrelated.

Does it happen when you install from the current binary? And what theme and system locale are you using, per chance?

If possible - I'm not sure what's the preference here - could you post this issue over at the actual EmulationStation tracker?

https://github.com/RetroPie/EmulationStation/issues

We'll follow up there as much as possible. It might be related to this prior problem, or it might be completely unrelated.

Thanks for raising this.

gaenserich, 2017-09-29:
@pjft I respond here since with your edit you seem to think it may be relevant to this particular bug. Should I also post to the ES tracker, too (and thus duplicate the issue) or keep building this thread?

> Just to be clear, though, you're getting this error on a _Pi_?

Yes. A RPi 3.

> Does it happen when you install from the current binary?

Yes. I updated from binary yesterday, and its still throws this error.

> And what theme and system locale are you using, per chance?

Right now the Tronkyfran theme, but it shows the exact same error with default Carbon.

`locale`:
```sh
LANG=en_US.UTF-8
LANGUAGE=
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=
```
It occurs to me that not many English-speaking users would switch the locale from en_UK.UTF-8 (the default, correct?) Perhaps switching back to the default might shed some light on this situation.

pjft, 2017-09-29:
The main reason I was suggesting to post in the ES tracker is because this bug has nothing to do with RetroPie-Setup, which is where this original bug had been mistakenly posted. As such, the wider RetroPie ES developer community won't be notified - or be made aware of it.

I don't think it is related to the original bug here. This was only happening on Linux, not on a Pi. And, as I said, the code that fixed the previous problem is still in place.

To expand on the issue, this usually happens because of some sort of mis-handling of a char* pointer. At the time, the instance that was causing this error on Linux was tackled, that's why I'm suggesting that while there may be the chance that it is related, it might be a completely new instance of it. 

The locale question was more to understand whether you'd be in a non-US/UK locale, which doesn't seem to be the case. Same goes for the them question.

Thanks.

I suppose I'll need to start asking more troubleshooting questions, then, as this doesn't seem to be extremely common, especially on a Pi.

Could you share the content of your es_settings.cfg file, and the last es_log.txt (or es_log.txt.bak) file that you'd have, from a session that crashed? Maybe those will share some insights into what the problem may be.

Also could you, if you have the chance, make a video of it happening, from the moment you start ES until it crashes, and not navigate especially quickly (so that in the video we can see what's happening on screen), and send it over? 

Sorry for the trouble, and thanks for attempting this. 

I don't personally mind continuing the conversation here, especially as I don't know what the best practice/recommended approach for this is. I still feel it would be useful to continue in the ES tracker, even if your first post just summarizes the issue and contains a link here, and then the follow up answer.

Your call, at least until we get more formal guidance from others.

Thanks.