RetroPie/RetroPie-Setup #2097: Openbor source

himura95, 2017-08-23:
Hello,

I see in RetroPie-Setup/scriptmodules/ports/openbor.sh

function sources_openbor() {
    gitPullOrClone "$md_build" https://github.com/rofl0r/openbor.git
}
This repo is dead in 2013, is it possible to modify with the official and still updated repo of Openbor Project ? :

function sources_openbor() {
    gitPullOrClone "$md_build" https://github.com/DCurrent/openbor.git
}

Thanks.

Darknior, 2017-08-23:
WOAW it can be a dream if one day it is possible to compile the last version on PI and use it <3
The last mods are really fantastic and better than  the original arcade and console games !
But i'm not a coder and i can't help for this :(

joolswills, 2017-08-23:
@himura95  - There is already a ticket regarding openbor issues. https://github.com/RetroPie/RetroPie-Setup/issues/1637 - you should post to the forum first as clearly detailed when opening an issue.

@Darknior this is the wrong place for comments like this, please use the forum for discussion.

closed 2017-08-23 by joolswills