RetroPie/RetroPie-Setup #2109: ARM64 support?

raidensnake, 2017-09-10:
I've been trying to see if it's remotely possible to compile retropie on arm64. I've recently discovered that the newest version of the kernel and userland with a few minor tweaks allows it to add the full hardware support. I've managed to run tests by recompiling a game and managed to boot it. however I get errors trying to boot emulationstation amongst some builder issues like -mfpu and -marm not being supported in the compilers. I'm wondering of anyone is interested in looking into this.

joolswills, 2017-09-10:
You should use the forum for discussion as clearly mentioned when opening an issue

>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

>For ideas and enhancements, please also post to the forum first, to see if it is something we would consider and to see if other users would find it useful.

We have no plans for aarch64 support for the RPI, and for emulators that have armv7/neon optimisations, performance would be worse.

closed 2017-09-10 by joolswills

raidensnake, 2017-09-10:
The forum topics were closed as I was told to post the request here and not there.

joolswills, 2017-09-10:
Please point me to where someone said that.

raidensnake, 2017-09-24:
I would except the topic got removed.

joolswills, 2017-09-24:
No it didn't.