RetroPie/RetroPie-Setup #2110: dgen does not build from source

ProxyCell, 2017-09-10:
Version: latest pull from today (Sept 10, 2017)
First noticed: earlier this year, but only used a "fresh" 4.2 RetroPie image today, then updated to the latest repo and tried to built dgen from source
RPi: 3
[rps_2017-09-09_221204.txt](https://github.com/RetroPie/RetroPie-Setup/files/1290507/rps_2017-09-09_221204.txt)

I have attached a log and I also tried building it from source a 2nd time afterwards.
[rps_2017-09-09_221525.txt](https://github.com/RetroPie/RetroPie-Setup/files/1290508/rps_2017-09-09_221525.txt)

joolswills, 2017-09-10:
Thanks. Surprised anyone is still using this emulator on retropie. Will apply a quick fix (same as we do for mame4all).

closed 2017-09-10 by joolswills

ProxyCell, 2017-09-10:
I like to keep all controller-capable emulators installed. thank you for the fix, it works here.

commit 6ea7f002 (2017-09-10): dgen - fix compilation on rpi3 - fixes #2110