RetroPie/RetroPie-Setup #2111: Add CHD support for Sega CD / TurboGrafx-CD / SuperGrafx

msheehan79, 2017-09-11:
Recent updates to the lr-beetle-pce-fast, lr-beetle-supergrafx and lr-genesis-plus-gx cores added CHD support. Can these new extensions be added to the platforms.cfg for these systems and added to the wiki?

joolswills, 2017-09-13:
Please provide more details - which extensions ? is it just `.chd` ?

msheehan79, 2017-09-13:
Yes, just .chd (and .CHD) formats are now supported. 

There are some discussions in the various libretro repositories:
https://github.com/libretro/Genesis-Plus-GX/issues/93
https://github.com/libretro/beetle-supergrafx-libretro/issues/22
https://github.com/libretro/beetle-pce-fast-libretro/issues/62

In terms of space savings its pretty significant, finding about 50% compression vs. the standard bin/cue format, plus now everything is stored in 1 file instead of 2.

A useful post for users looking for info on how to convert their bin/cue files to CHD is on the Libretro forums: https://forums.libretro.com/t/chd-batch-for-newly-supported-systems/11673

You can find the chdman utility to convert these files included with the latest MAME distributions.

Darknior, 2017-09-13:
Woaw great news :D 
They write too **Beetle Saturn** ... it is not working fine today on PI but one day with PI XX it will ;)
I don't found if it is also available for other systems using ISO like 3DO.
So bad it don't permit to MERDE Multi-CD games like PBP on PSX :(

joolswills, 2017-09-17:
fixed in 0a3faa9d6f996ba8581f909aa0f7e250400e3543 (I put wrong ticket # in the commit msg)

closed 2017-09-17 by joolswills