RetroPie/RetroPie-Setup #2121: Minor: retroarch-xmb-monochrome.tar.gz needs updating

psyke83, 2017-09-17:
Hi,

It seems that the retroarch-xmb-monochrome.tar.gz archive is out out date. From a quick perusing of the menus with the xmb menu active, the "Add to Favorites" menu item icon and the Favorites menu icon are missing.

Manually installing the full retroarch assets can alleviate the issue, but it's probably a good idea to update the archive.

I can see the missing icons in recent commits: https://github.com/libretro/retroarch-assets/commits/master/xmb/monochrome

joolswills, 2017-09-18:
Thanks. Will update.

psyke83, 2018-12-21:
Jools,

I just performed a full update of all packages and noticed missing icons again. Rather than me or others pestering you each time this happens, and to avoid you having to remember to manually perform this step for generated binaries, could we perhaps take steps to automate this process?

Would this be sufficient? https://github.com/RetroPie/RetroPie-Setup/compare/master...psyke83:auto_xmb_assets

I'm unsure if this is sufficient to solve the issue, as I don't know if your chroot's `$__tmpdir/archives/` actually corresponds to `$__archive_url`. If not, you might need to do an rsync too?

Also, if you get annoyed having to clone the assets repository on each build, don't forget that the function could be changed to use the much more efficient `svn export` to minimize bandwidth usage/time.

Not sending this as a PR as I would expect if you want to do this, you'd prefer to deal it this yourself (assuming the above change is not sufficient). Thanks & sorry for the trouble :)

joolswills, 2018-12-21:
Sorry thought I had done this with last retroarch build. Might consider automating it, but maybe via builder script. It's done now anyway. Thanks for heads up.

closed 2018-12-21 by joolswills