RetroPie/RetroPie-Setup #2128: Add lr-atari800 core

bigboo3000, 2017-09-23:
Hi
The atari800 libretro core is now more complete and can run games for both a5200 and a800.
Could you please add it to Retropie?

Darknior, 2017-09-23:
WOAW so excellent <3

cpoliticas, 2017-09-23:
The emulator is in the experimental packages, but for some reason the installation is broken

joolswills, 2017-09-24:
It's not in experimental yet. I have an unfinished module and have written on the forum that I would complete/test it after the 4.3 release.

Please follow the guidelines of the issue tracker when opening a ticket

>For ideas and enhancements, please also post to the forum first, to see if it is something we would consider and to see if other users would find it useful.

psyke83, 2017-11-25:
Resolved via commit cd3869ae98b7664ca9f1a28190b6f75376c64665.

closed 2017-11-25 by psyke83