RetroPie/RetroPie-Setup #2131: [Patch] Add support for LinuxMint DE2

lexa2, 2017-09-23:
Please apply the following patch to add support for building RetroPie on LMDE2 (debian-based LinuxMint):
```
diff --git a/scriptmodules/system.sh b/scriptmodules/system.sh
index 623ff34..ae5def4 100644
--- a/scriptmodules/system.sh
+++ b/scriptmodules/system.sh
@@ -106,7 +106,14 @@ function get_os_version() {
             ;;
         LinuxMint)
             if compareVersions "$__os_release" lt 17; then
-                error="You need Linux Mint 17 or newer"
+                if [[ "$__os_codename" == "betsy" ]] ; then
+                    __os_codename="jessie"
+                    __os_debian_ver="8"
+                    __os_release="8.1"
+                    __os_id="Debian"
+                else
+                    error="You need Linux Mint 17 or newer"
+                fi
```

joolswills, 2017-09-24:
Not sure I want to include this as is sorry. Will check if there is a better way than having to set all the os globals based on the codename like that or whether we can do this more automatically.

lexa2, 2017-09-24:
@joolswills Sure, feel free to modify this patch any way you would see fit. It is just a quick-n-dirty change I used for my child's box to get setup script to work on LMDE2. For now people searching for help would be able to use this patch from the comment above while you would be able to figure out the solution that fits you best.

commit 5a120b89 (2017-12-28): system - LinuxMint DE system identification - fixes #2131
closed 2017-12-28 by joolswills