RetroPie/RetroPie-Setup #2132: Installer missed to put a link to retroarch into /usr/bin on Linux Mint

KarlZeilhofer, 2017-09-27:
I've just installed RetroPi on a Linux Mint 18.1 system. The installer worked smoothly. But I couldn't start any ROM. 

With this hint here: 
https://retropie.org.uk/docs/FAQ/#how-would-i-start-from-command-line-say-the-snes-emulator-by-itself
I found out, that `retroarch` isn't in my search path. 

The solution was to make a symlink from `/opt/retropie/emulators/retroarch/bin/retroarch` to `/usr/bin
/`

joolswills, 2017-09-27:
You ignored the instructions when opening a ticket - 

>This issue tracker is NOT for Help / Support - Please use the RetroPie forum - https://retropie.org.uk/forum/

>Even if you think you have a bug please post to the forum first, as many "issues" can be solved with configuration changes.

This isn't a bug by by design - only Emulation Station has a pathed binary for launching the frontend. Everything else is accessed directly from the installed location via runcommand.

joolswills, 2017-09-27:
The docs need updating in this case as they should include the full path to retroarch - the docs are user edited, so whoever added that forgot to include the full path.

closed 2017-09-27 by joolswills