RetroPie/RetroPie-Setup #2151: [RPi3] Exception OSError

LodanZark, 2017-10-11:
In the terminal, after leaving the retropie_setup.sh, I got this message:
`Exception OSError: (9, 'Bad file descriptor') in <module 'threading' from '/usr/lib/python2.7/threading.pyc'> ignored
`
I'm not sure if it is a result of a recent commit or if it's caused for some faulty config on my side

joolswills, 2017-10-11:
You have provided no information about your set-up - Also please use the forum first for reports / support as detailed when opening a ticket - the forum also provides information of what is required (I'm sure I have mentioned this before to you).

closed 2017-10-11 by joolswills

joolswills, 2017-10-11:
I managed to reproduce this on my rpi3

@psyke83 Looks like your commit did it https://github.com/RetroPie/RetroPie-Setup/pull/2148

It's a non fatal error, but I will revert your change for now.

psyke83, 2017-10-11:
Yes, it was a sys.exit() exception likely caused by repeated calls of signal_handler. Increasing the sleep in the loop would help but no guarantee it would stop this in all cases.

I also tried to make joy2key.py instead pass a signal back to the parent bash process which joy2KeyStart could wait to receive, but didn't have much luck yet. Best to just leave this alone for now as it's just wasting your time.

joolswills, 2017-10-11:
I don't like the fact that joy2key is started for commandline stuff anyway - so I think I will change it so it's only started when needed (via retropie-setup or the retropie menu launch module etc)

joolswills, 2017-10-11:
That should mean there is enough delay etc.