RetroPie/RetroPie-Setup #2154: ".7z" and ".7Z" Entries Missing for the Following Systems with 7-Zip Compression Supported Emulators in RetroPie's Auto-Generated "es_systems.cfg"

Eckaji, 2017-10-12:
All systems below have had their "lr-" emulators tested for ".7z" support and they do indeed have support, barring the libretro MAME emulators in Arcade.

- Arcade
- Final Burn Alpha
- Famicom Disk System
- Neo Geo
- Neo Geo Pocket
- Neo Geo Pocket Color
- Sega SG-1000

It would also be nice if the extensions were alphabetized for easier reading, but I know that isn't a true issue.

Hopefully I have done this right, since this did not seem like something that needed to be discussed. If I've done something wrong, then I apologize in advance.

Pundan, 2017-10-17:
In your es_systems.cfg file you can change this. It's just a matter of adding .7z as a file extension that can be run for each system. See this guide on how to edit the es_systems.cfg, it's recommended to make a copy of it. [https://github.com/RetroPie/RetroPie-Setup/wiki/Add-a-New-System-in-EmulationStation](url)

Eckaji, 2017-10-17:
I know how to do that and have done so. My issue is that it isn't done by default, even though all of the emulators above support 7z, and this should be done so that every user looking to save up to 15% more space doesn't have to.

commit 9b709900 (2017-10-17): platforms.cfg - added .7z for some additional platforms - fixes #2154
closed 2017-10-17 by joolswills