RetroPie/RetroPie-Setup #2157: ".bs" and ".BS" Entries Missing for SNES Entry in RetroPie's Auto-Generated "es_systems.cfg"

Eckaji, 2017-10-13:
I'm not sure what else to add beyond the title, other than that I tested Satellaview emulation and it does work, though Satellaview ROMs have to be in an uncompressed state with the ".bs" extension to be playable.

Pundan, 2017-10-17:
You need to add the .BS file extension to your SNES-config in your es_systems.cfg in order for the ROMs to show up. [https://github.com/RetroPie/RetroPie-Setup/wiki/Add-a-New-System-in-EmulationStation](url)

Eckaji, 2017-10-17:
Yes, I know. 

My issue is that this isn't the default, as most Satellaview ROMs distributed nowadays have the ".bs" extension, and this would save the end-user from having to do so.

Still, I believe the added support of compression by just changing the ".bs" every ROMs extension to ".sfc" is the best route for most users... and I should go add a small section on Satellaview in the SNES part of the wiki advising to do so.

joolswills, 2017-10-17:
Note you don't need to edit `es_systems.cfg` to add additional extensions (and they will get ovewritten).

Just create a `platforms.cfg` file in `/opt/retropie/configs/all` with 

```
snes_exts=".7z .bin .bs .smc .sfc .fig .swc .mgd .zip"
snes_fullname="Super Nintendo"
```

and then update any snes emulator. you can override any extensions like this (and system names / themes)

joolswills, 2017-10-17:
I will add this to the system platform.cfg anyway, but if you want to ever adjust other systems extensions use the method above.

commit 02ba5817 (2017-10-17): platforms.cfg - added .bs extension for snes - fixes #2157
closed 2017-10-17 by joolswills