RetroPie/RetroPie-Setup #2165: use default group when installing/updating packages

mioux, 2017-10-19:
On raspberry pi, when installing or updating packages, retropie_setup.sh considers it exits a group with the same name as the user. After installation almost all of the retropie files are owned by root:root

*Step to reproduce :*
Install Raspbian lite
Run these commands

    $ usermod -U root
    $ sudo passwd
    $ exit
    __relog in root__
    # usermod -l newlogin
    # exit
    __relog in newlogin__
    $ cd path/of/retropie/setup
    $ sudo ./retropie_setup.sh

joolswills, 2017-10-19:
It is currently a requirement that you have a user with a group of the same name. This is unlikely to be changed - user adduser, or set-up a group manually.

closed 2017-10-19 by joolswills

mioux, 2017-10-21:
Can I propose you a patch to check if the users belongs to a group with the same name ?

I also have made all errors message go to standard error instead of standard output in retroarch_packages.sh

[check-group.txt](https://github.com/RetroPie/RetroPie-Setup/files/1404404/check-group.txt)