RetroPie/RetroPie-Setup #2171: line 1006: 7693 Segmentation Fault when running np2pi

sxales, 2017-10-23:
I've installed np2pi from the experimental section on my RPI 3 with RetroPie 4.3.3 but when I try to run anything I get the above error in the runcommand.log at /dev/shm. I am sure is an upstream problem but since there is no issue tracker for np2pi, here is as goo d as anywhere else.

joolswills, 2017-10-23:
You should wait for help on the forum first as detailed when opening an issue.

sxales, 2017-10-23:
Granted 12 hours isn't very long to wait but when I've posted on the forum about np2pi problems before I've never gotten a response. So, 12 hours with no response seemed like a pretty good indication that was going to happen again.

joolswills, 2017-10-23:
This is not a full time job. Please follow our guidelines.

closed 2017-10-23 by joolswills

sxales, 2017-10-23:
Fair enough, at what point should I re-post this when no one responds?