RetroPie/RetroPie-Setup #2180: Bug in the es_input.cfg on Ubuntu x86.

Dragontorc, 2017-11-01:
es_input.cfg is trying to access inputconfiguration.sh in the following directory: /opt/retropie/supplementary/emulationstation/scripts/

This path doesn't exist.  Instead of /emulationstation there's /emulationstation-dev folder.  As a result, ES is asks to configure input  each time is started (configured input settings are never written to the es_input.cfg).

joolswills, 2017-11-01:
For now please configure the controller on emulationstation, before installing emulationstation-dev

Dragontorc, 2017-11-01:
I did exactly that.  However, each time I restarted ES it asked me to configure input again.  I saw the message in the console that es_input.cfg couldn't find:
/opt/retropie/supplementary/emulationstation/scripts/inputconfiguration.sh
When I 've looked for that directory, I've found inputconfiguration.sh
 in the  /opt/retropie/supplementary/emulationstation-dev/scripts/ instead. 
I've changed path in the es_input.cfg file to /opt/retropie/supplementary/emulationstation-dev/scripts/ then restarted ES and configured input.  After that, ES doesn't ask me anymore to configure input each time is restarted.

I've installed Retropie according to the tutorial on the RetroPie wiki (with all core, main, optional and experimental packages installed from the sources).  I did that on 3 different PCs and all have same issue.  All PCs run latest Ubuntu 17.10.  Is it possible that somehow I got wrong version (some sort of development) of RetroPie installed and hence /emulationstation-dev directory shows up instead of /emulationstation?

Thank you for looking into it.

joolswills, 2017-11-01:
You should not install everything really - especially from experimental - it's called experimental for a reason.

joolswills, 2017-11-01:
The correct install is to use the "basic install" from the main menu.

Dragontorc, 2017-11-01:
I did basic install first, then the rest of the packages.  I just have looked up in the installer and see that Emulationstation (item 810) doesn't show INSTALLED at all.  Trying to install it now.

joolswills, 2017-11-01:
That's because emulationstation-dev removes emulationstation.

Dragontorc, 2017-11-01:
I see.  I have successfully reinstalled 810 Emulationstation and ES doesn't ask for input configuration any more.
Thank you, appreciate your help.

closed 2017-11-01 by joolswills