RetroPie/RetroPie-Setup #2188: Disconnected bluetooth controller causes freezes on regular intervals

initialize, 2017-11-05:
Hardware: Raspberry Pi 3
Retropie version: 4.3.3
Emulation station: 3.6.5RP
Controller: 8bitdo fc30 pro
Kodi: 17.4

Description:

When bluetooth paired controller powers off, bluetoothd probes for a reconnect every ten or so seconds. When doing so, the raspberry pi eezes for approximately a second. This becomes an issue when working on the hardware over ssh or streaming media in Kodi.

Reproduce:

1. Connect a bluetooth controller
2. Power off bluetooth controller
3. ssh to device
4. Keep command line busy
5. Notice freezes

From /var/log/syslog
```
Nov  1 00:13:03 retropie bluetoothd[777]: Can't get HIDP connection info
Nov  1 00:13:03 retropie rsyslogd-2007: action 'action 17' suspended, next retry is Wed Nov  1 00:13:33 2017 [try http://www.rsyslog.com/e/2007 ]
Nov  1 00:13:08 retropie bluetoothd[777]: connect error: Host is down (112)
Nov  1 00:13:13 retropie bluetoothd[777]: a2dp-sink profile connect failed for <BT ADDRESS>: Protocol not available
Nov  1 00:13:33 retropie bluetoothd[777]: Can't get HIDP connection info
Nov  1 00:13:33 retropie rsyslogd-2007: action 'action 17' suspended, next retry is Wed Nov  1 00:14:03 2017 [try http://www.rsyslog.com/e/2007 ]
Nov  1 00:13:38 retropie bluetoothd[777]: connect error: Host is down (112)
Nov  1 00:13:43 retropie bluetoothd[777]: a2dp-sink profile connect failed for <BT ADDRESS>: Protocol not available
Nov  1 00:14:04 retropie bluetoothd[777]: Can't get HIDP connection info
Nov  1 00:14:04 retropie rsyslogd-2007: action 'action 17' suspended, next retry is Wed Nov  1 00:14:34 2017 [try http://www.rsyslog.com/e/2007 ]
Nov  1 00:14:09 retropie bluetoothd[777]: connect error: Host is down (112)
Nov  1 00:14:14 retropie bluetoothd[777]: a2dp-sink profile connect failed for <BT ADDRESS>: Protocol not available
Nov  1 00:14:34 retropie bluetoothd[777]: Can't get HIDP connection info
```

ps -aef|grep blue
```
root       777     1  0 00:03 ?        00:00:00 /usr/lib/bluetooth/bluetoothd
root       880     1  0 00:03 ?        00:00:00 /bin/sh -c while true; do for mac in <BT ADDRESS> <BT ADDRESS> <BT ADDRESS>; do hcitool con | grep -q "$mac" || echo "connect $mac" | bluetoothctl >/dev/null 2>&1; sleep 10; done; done
```

Cheers

joolswills, 2017-11-06:
You should use the forum as described when opening an issue. Also turn off background connect mode in Bluetooth settings. It's a hack that shouldn't be used generally.

closed 2017-11-06 by joolswills

initialize, 2017-11-06:
I did, one week ago. Since there were no replies about resolution, I'd say it is a safe assumption that it is a bug that should be tracked.