RetroPie/RetroPie-Setup #2189: Update error: Package sdl1 '404 Not found'

angrystar170, 2017-11-06:
please, check the log at the bottom and fix it. 
==== LOG ====
Log started at: 2017. 11. 06. (월) 22:25:21 KST

RetroPie-Setup version: 4.3.4 (99bdad7)
System: Linux retropie 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux

= = = = = = = = = = = = = = = = = = = = =
Installing dependencies for 'advmame-0.94' : AdvanceMAME v0.94.0
= = = = = = = = = = = = = = = = = = = = =

Did not find needed package(s): libsdl1.2-dev. I am trying to install them now.

= = = = = = = = = = = = = = = = = = = = =
Installing 'sdl1' : SDL 1.2.15 with rpi fixes and dispmanx
= = = = = = = = = = = = = = = = = = = = =

--2017-11-06 22:25:24--  https://files.retropie.org.uk/binaries/jessie/rpi3/libsdl1.2debian_1.2.15-11~rpi1~jessie_armhf.deb
Resolving files.retropie.org.uk (files.retropie.org.uk)... 93.93.129.253, 2a00:1098:0:80:1000:57:0:1
Connecting to files.retropie.org.uk (files.retropie.org.uk)|93.93.129.253|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2017-11-06 22:25:26 ERROR 404: Not Found.

--2017-11-06 22:25:26--  https://files.retropie.org.uk/binaries/jessie/rpi3/libsdl1.2-dev_1.2.15-11~rpi1~jessie_armhf.deb
Resolving files.retropie.org.uk (files.retropie.org.uk)... 93.93.129.253, 2a00:1098:0:80:1000:57:0:1
Connecting to files.retropie.org.uk (files.retropie.org.uk)|93.93.129.253|:443... connected.
HTTP request sent, awaiting response... 404 Not Found
2017-11-06 22:25:28 ERROR 404: Not Found.

joolswills, 2017-11-06:
@psyke83 - must be after your SDL changes. I'm going to revert for now.

joolswills, 2017-11-06:
Aah. I need to rebuild binaries after the SDL changes. Didn't realise the binary packages had been renamed.

joolswills, 2017-11-06:
As I don't have time right now I'm still going to revert. Not sure I want the existing packages renamed really.

psyke83, 2017-11-06:
Sure, whatever's best for you. I figured that adding the codename might be useful since you're supporting two distro versions, and the tilde is useful so that an upstream update to the same version will be marked as newer; your packages will still be held, but at least you will see the "held back" in apt-get upgrade to help you realize if/when an update is needed.

joolswills, 2017-11-06:
we use separate folders for the binaries on the server so technically we don't need the distro name in the packages (and we don't do it for anything else).

joolswills, 2017-11-06:
This issue is resolved - I will revisit the sdl1 on stretch when I am building the bins for stretch etc.

closed 2017-11-06 by joolswills