RetroPie/RetroPie-Setup #2207: GlideN64 broken

cptbananas, 2017-11-21:
A recent change to the GlideN64 plugin for Mupen64plus (NOT lr version) resulted in all games becoming unplayable for this plugin. The change is presumed to have entered the repository between 11/1 and 11/17/2017. Previously, this was regarded by many as the best plugin for N64 emulation on Retropie.

Confirmation of multiple users experiencing the issue is discussed in this thread:
https://retropie.org.uk/forum/topic/14448/n64-emulators-not-loading-textures/1

To replicate:
-Update mupen64plus, either from binary or from source
-Load a rom in mupen64plus with GlideN64 plugin (gles2rice is not affected)

You will encounter the message "init noise textures," followed by grossly broken graphical rendering. See screenshots of "Super Mario 64" and "Mario Kart 64" as examples.

[image]
[image]

joolswills, 2017-11-21:
If upstream come up with the fix shortly, I will rebuild binaries on the latest code. if not, I will consider bisecting to see when the problem started and roll back the code for the RPI until it's fixed.

johnmanko, 2017-11-23:
@cptbananas / @joolswills GLideN64 is deferring to RetroPie as a Pi GPU driver issue.

dankcushions, 2017-11-23:
@johnmanko not deferring, as such. plenty of GPU drivers have issues. it’s just an observation. it will still be fixable within GLideN64.

psyke83, 2017-12-23:
This is already resolved despite upstream's bug not yet being closed, so I'll close ours. When we move to stretch, there's a segmentation fault related to the shader cache that will need to be fixed, but it's not related to the missing textures anyway.

closed 2017-12-23 by psyke83

johnmanko, 2017-12-23:
when can we see this available to RetroPie installations?

joolswills, 2017-12-23:
It already is. Please use the forum for help.