RetroPie/RetroPie-Setup #2227: Add scummvm-2.0.0 support

dafioram, 2017-12-21:
Scummvm 2.0.0 is out.

Users can pull 2.0.0 from source and compile just fine with scummvm.sh
This can be accomplished by changing:
RetroPie-Setup/scriptmodules/emulator/scummvm.sh
line #30
-gitPullOrClone "$md_build" https://github.com/scummvm/scummvm.git "branch-1-9"
+gitPullOrClone "$md_build" https://github.com/scummvm/scummvm.git "branch-2-0-0"

2.0.0 was developed with sdl 2.0 in mind so I am not sure how well compiling from source for sdl1 will work (i.e., scummvm-sdl1.sh).

Currently, the raspberry pi binary for scummvm 2.0.0 on scummvm's website cannot not be downloaded and used on retropie (specifically a raspberry pi) because it was compiled with a higher png library version than raspian jessie has (it was compiled with libpng16 and jessie only has libpng12). So if a binary scummvm 2.0.0 is to be used on a raspberry pi running retropie then either:
1. Retropie itself must compile libpng16 from source (maybe someone could upload an unofficial libpng16 binary for jessie).
2. We wait till Retropie is running on raspian stretch (which has libpng16 standard).
3. Someone compiles scummvm 2.0.0 on retropie 4.3 (jessie) by changing the line of code as shown above to pull from 2-0-0 instead of 1-9 and then upload this binary somewhere that other retropie users can grab. This method will compile scummvm 2.0.0 with libpng12 so it will work on other retropie's running 4.3.

joolswills, 2017-12-22:
Thanks - We are planning to update - we were just making sure everything works. The sdl1 module will be removed as it was added due to some joystick bugs in 1.9 with sdl2.

closed 2018-01-19 by joolswills

AnthraxLeprosyPi, 2018-02-19:
Hey guys,

Is there any update on this?
If I understand the above correctly I could just build from source and it would work on a raspberry pi 3?

Best regards 
Anthrax

Darknior, 2018-02-19:
Read before : https://github.com/RetroPie/RetroPie-Setup/commit/f747616e5599f0d8824b9f4900da25bfc4586813

Yes you can build from source the 2.0

AnthraxLeprosyPi, 2018-02-19:
Ah, nice thx you guys!

joolswills, 2018-02-19:
Please use the forum for any further questions.

AnthraxLeprosyPi, 2018-02-19:
Worked like a charm. 

Forums next time!

commit f747616e (2018-01-19): scummvm - update to 2.x branch. closes #2227
 * scummvm built against sdl2 now has fixed joypad support, so the scummvm-sdl1 is no longer needed. However it is being left for now, in case there are other isues.