RetroPie/RetroPie-Setup #2241: symlink for cannonball custom music files

ProxyCell, 2017-12-28:
Hi, can the symlink for cannonball's custom music files be added?

Source: https://retropie.org.uk/forum/post/118254

"/opt/retropie/ports/cannonball/custom_music" - I guess this would need a real folder in another location like "/home/pi/RetroPie/roms/ports/cannonball/custom_music/" ?

joolswills, 2017-12-28:
I don't know whether that would work - it was an idea, but I can test.

cmitu, 2020-10-25:
By specifying the full path to the custom WAV files, there's no need for a symlink. Users can create a `music` sub-folder and copy the custom track files there, then add the full path in `config.xml`.

Clarified with an example in https://github.com/RetroPie/RetroPie-Docs/commit/d89ebde30dbcc53ffd7c981e7d7b3ab2bf2fc09b.

closed 2022-12-30 by cmitu