RetroPie/RetroPie-Setup #2249: ppsspp-lr does not build on ODROID - libavformat/avformat.h: No such file or directory

Kaeltis, 2018-01-02:
Trying to build lr-ppsspp (or ppsspp) on ODROID XU4 with Ubuntu 16.04.3 results in an error:

```
../Core/HLE/sceAtrac.cpp:94:34: fatal error: libavformat/avformat.h: No such file or directory
compilation terminated.
Makefile:339: recipe for target '../Core/HLE/sceAtrac.o' failed
make: *** [../Core/HLE/sceAtrac.o] Error 1
```
Full log: http://termbin.com/hrww

Apparently this issue existed for i486 as well and was fixed there some time ago: https://github.com/libretro-mirrors/libretro-ppsspp/issues/73

zerojay, 2018-01-02:
So I went down this rabbit hole a little while ago: https://retropie.org.uk/forum/topic/15375/adventures-with-odroid-xu4-and-retropie

I concluded that if you install libavformat-dev and libswscale-dev (which will be the next error you hit if you just installed libavformat-dev to satisfy the above issue), when you build, you will get a build of lr-ppsspp that works except all ffmpeg decoding would be broken. So taking a look at the thread I posted above might be helpful in figuring out what needs to be done here.

joolswills, 2018-01-02:
it definitely built before, so I will look into it. Not sure I want to use the system ffmpeg headers, as we don't on other platforms (and there is a risk of a mismatch happening with future versions). I will check when I have a chance.

psyke83, 2018-01-02:
The current lr-ppsspp repo is probably going to become obsolete soon - there's a PR on the upstream project to integrate libretro directly.

There are still issues with ppsspp on Pi related to build errors from Wayland/Vulkan, and not setting a fullscreen context. I've suggested potential fixes for some - still waiting for feedback.

joolswills, 2018-01-02:
Thanks.

I may move temporarily to our fork of the lr-ppsspp then (which we use for the rpi), but of course we should move to the newer libretro ppsspp repo in the future.

MJB-G, 2018-05-06:
Hi

Still impossible to build : 

on ODROID XU4 with Ubuntu 18.04

Makefile:316: recipe for target '../Core/HLE/sceMpeg.o' failed
make: *** [../Core/HLE/sceMpeg.o] Error 1
make: *** Attente des tâches non terminées....
make : on quitte le répertoire « /home/odroid/RetroPie-Setup/tmp/build/lr-ppsspp/libretro »
~/RetroPie-Setup
Could not successfully build lr-ppsspp - PlayStation Portable emu - PPSSPP port for libretro (/home/odroid/RetroPie-Setup/tmp/build/lr-ppsspp/libretro/ppsspp_libretro.so not found).

steeeb, 2018-10-18:
Similar problems on Tinkerboard too by the way (Exact same file)