RetroPie/RetroPie-Setup #2266: PSP Snes gamepad issues

hodlenn, 2018-01-22:
I'm newer to RetroPie and RetroArch, so please excuse my inexperience.

I am running RetroPie 4.3.9 on a Raspberry Pi 3 with a 32 GB SD card. Currently, I am having an issue with the PSP system (Monster Hunter 3, specifically). I am using the iBuffalo Classic USB Gamepad, and when I boot up the game, I am not given any warning/yellow text, but some of the buttons do not function properly. When I use the upper and lower arrows on the Dpad, they do not have any function. Does anyone know what I can do to fix this? Thank you!

joolswills, 2018-01-23:
Please use the forum for support - https://retropie.org.uk/forum/

closed 2018-01-23 by joolswills

hholst80, 2018-02-10:
@hodlenn I seem to have a similar issue with my controller can you reference your support post when if you made one?